oneapi-src/oneAPI-samples #56: question about __attribute__((always_inline)

zjin-lcf, 2020-08-06:
I don't observe performance difference whether __attribute__((always_inline)) is added for the sepia filter. 
Can you please explain if such attribute is needed ? 

Thanks

// always_inline as calls are expensive on Gen GPU.

__attribute__((always_inline)) static void ApplyFilter(uint8_t *src_image, 
                                                       uint8_t *dst_image,
                                                       int i) {

pmpeter1, 2020-08-06:
This is the default on Gen GPU, and currently can not be turned off so therefore it's not needed.

zjin-lcf, 2020-08-06:
Sorry I don't understand what you said. Do you mean that a function called in a SYCL kernel is, by default, inlined ?

pmpeter1, 2020-08-06:
Yes, on Gen ALL function calls are lined during compilation.   In the future this might be changed, but that is how it works today.

closed 2020-08-06 by zjin-lcf