oneapi-src/oneAPI-samples #63: Feature update for intrin_dot_sample.cpp

zjin-lcf, 2020-08-08:
# Summary
Include a short summary of the request. Sections below provide guidance on
what factors are considered important for a feature request.

README says that "They provide access to instructions that cannot be generated using the standard constructs of the C and C++ languages, and allow code to leverage performance enhancing features unique to specific processors. " 
May we use some SIMD compiler options to achieve vectorization  ? 

# Problem statement
A comparison between the performance of using intrinsics and the performance of using SIMD compiler options

# Preferred solution
Add a solution to the problem where the size is not a multiple of 8. Currently, it is 24.
Add a solution, if possible, to adding some SIMD compiler options to achieve vectorization

Thanks

anjgola, 2020-08-11:
@ethanhirsch I think you are the author of this sample.

JoeOster, 2020-08-24:
@ethanhirsch is working on this

closed 2020-08-25 by JoeOster