oneapi-src/oneAPI-samples #66: Unit of time in iso2dfd.cpp

LaurentPlagne, 2020-08-11:
Hi,
I think the unit of time in DirectProgramming/DPC++/StructuredGrids/iso2dfd_dpcpp/src/iso2dfd.cpp
is not correct. "ms" are displayed while it should be "s".

anjgola, 2020-08-11:
@Alberto-Villarreal I think you are the author of the sample.

Alberto-Villarreal, 2020-08-11:
@LaurentPlagne  Hi, thanks for pointing that out. I will proceed to fix.

LaurentPlagne, 2020-08-13:
Thank you, and thank you for this example which is very useful ! 

I wish that [code sample walkthrough](https://software.intel.com/content/www/us/en/develop/articles/code-sample-two-dimensional-finite-difference-wave-propagation-in-isotropic-media-iso2dfd.html) exist for the other examples. The iso3DFD example in particular.

BTW: do you know is some ND-array wrapper allowing for cleaner syntax (e.g. a(i,j) in place of a[i+nxj]) have been developed for DPCPP kernels ? I try to write a simple one but it seems that I need a deeper understanding of the internal design of buffer and accessors...

JoeOster, 2020-08-24:
This was addressed in [PR# 86](https://github.com/oneapi-src/oneAPI-samples/pull/86/files)

closed 2020-08-24 by JoeOster