oneapi-src/oneAPI-samples #125: [Change Request] super(TestModel, self).__init__()  in PyTorch_Hello_World.py

louie-tsai, 2020-08-28:
move a discussion from PR to this issue.
The request is from keryell

original code:
```
class TestModel(nn.Module):
    def __init__(self):
        super(TestModel, self).__init__()
```

Suggested codes:
```suggestion
        super().__init__()
```

_Originally posted by @keryell in https://github.com/oneapi-src/oneAPI-samples/pull/103#discussion_r478302102_

JoeOster, 2020-10-08:
@louie-tsai  Can we close this issue?

keryell, 2020-10-09:
I guess it should be up-streamed to Pytorch itself if the old Python coding style comes from there...

jingxu10, 2020-11-17:
Hi @keryell , we are following PyTorch coding style [here ](https://pytorch.org/tutorials/beginner/examples_nn/dynamic_net.html#sphx-glr-beginner-examples-nn-dynamic-net-py).

closed 2020-11-18 by JoeOster

keryell, 2020-11-18:
Perhaps people doing machine learning can not use human learning... :-(
But perhaps the PyTorch community is looking at the same time at the oneAPI C++ coding style to devise what a good C++ coding style is... ;-)