oneapi-src/oneAPI-samples #131: Runfail on CPU device

XunZhangIntel, 2020-09-01:
# Summary
Run fail on CPU device.

# Version
dpcpp --version
Intel(R) oneAPI DPC++ Compiler Pro 2021.1 (2020.8.0.0819)

# Environment
Linux

# Steps to reproduce
dpcpp -O3 -fsycl -std=c++17 -O2 -g -DNDEBUG -ltbb -lsycl -lmpi -o dpc_reduce
SYCL_DEVICE_TYPE=CPU ./dpc_reduce
Number of steps is 1000000
terminate called after throwing an instance of 'cl::sycl::runtime_error'
  what():  OpenCL API failed. OpenCL API returns: -5 (CL_OUT_OF_RESOURCES) -5 (CL_OUT_OF_RESOURCES)
Aborted (core dumped)

JoeOster, 2020-09-02:
@XunZhangIntel - Do you have a specific sample this error is occurring on?

XunZhangIntel, 2020-09-03:
@JoeOster It is dpc_reduce sample test.(DPC++/ParallelPatterns/Dpc_reduce)

JoeOster, 2020-09-03:
I will be working with Todd and Loc to get this resolved

JoeOster, 2020-09-03:
We have identified the trouble points: The functions named native3 and native4. To test just comment out that portion of the code.

We will have a resolution in the next few days

terdner, 2020-09-04:
If you run the sample on a CPU as your default device,  you may need to increase 
the memory allocation for openCL.  You can do this by setting an environment variable, 
"CL_CONFIG_CPU_FORCE_PRIVATE_MEM_SIZE=16MB

I was able to reproduce the bug without this setting, and then
set env variable, reran and it passed.  
I am checking code in now that should do this automatically.

XunZhangIntel, 2020-09-15:
@terdner And how to run this sample test on Windows platform? I try it as below and it has 'Segmentation fault' issue:

> dpcpp /EHsc    /nologo /MD /bigobj src/main.cpp /Fedpc_reduce /link tbb.lib /link impi.lib
> SYCL_DEVICE_TYPE=GPU ./dpc_reduce
Rank #0 runs on: XXX, uses device: Intel(R) Gen9
Number of steps is 1000000
Segmentation fault

Regards
-Xun Zhang

terdner, 2020-09-15:
Not intended to run on Windows.

JoeOster, 2020-10-07:
Can we close this issue?

terdner, 2020-10-07:
This sample was not intended to run on FPGA.   This is something we can investigate for future. 
I believe we can close this issue.

closed 2020-10-07 by JoeOster