oneapi-src/oneAPI-samples #158: Run fail on FPGA emulator device

XunZhangIntel, 2020-09-14:
Summary
Run fail on FPGA emulator device.

Version
dpcpp --version
Intel(R) oneAPI DPC++ Compiler Pro 2021.1 (2020.8.0.0827)

Environment
Linux

Steps to reproduce
dpcpp -O3 -fsycl -std=c++17 -O2 -g -DNDEBUG -ltbb -lsycl -lmpi src/main.cpp -o dpc_reduce
SYCL_DEVICE_TYPE="ACC" ./dpc_reduce
Intel(R) FPGA Emulation Device
Number of steps is 1000000
terminate called after throwing an instance of 'cl::sycl::runtime_error'
  what():  OpenCL API failed. OpenCL API returns: -5 (CL_OUT_OF_RESOURCES) -5 (CL_OUT_OF_RESOURCES)
Aborted (core dumped)

JoeOster, 2020-09-16:
@XunZhangIntel - This has been assigned to our FPGA team, we will update you as soon as we can

AlbyB, 2020-09-17:
@XunZhangIntel This looks to be the same error as issue #131, and commenting out the same native3 and native4 functions will allow you to run the code without this error. This should be resolved along with the other CPU error.

XunZhangIntel, 2020-09-24:
@AlbyB I try to run with setting 'CL_CONFIG_CPU_FORCE_PRIVATE_MEM_SIZE=16MB' environment varible as #131 , but still get runtime error message. I am not sure whether there is any W/A for it now.

AlbyB, 2020-10-03:
@XunZhangIntel  Sorry, I was referring to commenting out all the function calls for calc_pi_dpstd_native3/4 , ex on lines 713/720, 678/679. After commenting out the lines with those function calls, it worked for me.

closed 2020-10-30 by AlbyB