oneapi-src/oneAPI-samples #225: read-only accessor does not return const, it should be change to read-write accessor

arkhodamoradi, 2020-10-06:
# Summary
Defining a *read-only* accessor results in error when being used as a *const* object.
In this example:
https://github.com/oneapi-src/oneAPI-samples/blob/master/DirectProgramming/DPC%2B%2B/Jupyter/oneapi-essentials-training/02_DPCPP_Program_Structure/src/complex_mult_solution.cpp#L61-L70

*V1* and *V2* must be defined as *read-write* in order to use *complex_mul*.

# Environment
DevCloud, Jupyter notebook

# Steps to reproduce
Run the example code in tutorial.

praveenkk123, 2020-10-07:
Hi,
Thanks for pointing the issue. I ran the above said code on Intel devcloud and its working good with V1 and V2 accessors set as read_only and V3 accessor set as write_only.

Please also look at the Complex.hpp (in the src folder) function where complex_mul function is returning a const object 
"Complex2 complex_mul(const Complex2& obj) const"

Can you please try on devcloud, uncomment specified code in step 1, step2 and step 3 and try again.

JoeOster, 2020-11-16:
@arkhodamoradi -any issues with closing this?

closed 2020-11-18 by JoeOster