oneapi-src/oneAPI-samples #232: Request to Add a oneAPI Sample

xmnboy, 2020-10-08:
# Description
Three samples to demonstrate the use of the dpct migration tool. These samples have been approved and developed in concert with Tom L., Yury P. and Swapna D.

# Dependencies
These samples have no external header or library dependencies that are necessary to be available within this repo. Obviously, they do require installation of the oneAPI dpct tool.

# Sample Folder
These three samples will be going into the `Tools/Migration` folder (the `Migration` folder is new). 

I should be issuing a pull request soon after this issue is submitted, assuming no problems. :smile:

JoeOster, 2020-10-08:
I see no issue with the proposed samples or folder location

JoeOster, 2020-10-08:
Please submit your PR

xmnboy, 2020-10-08:
@JoeOster -- PR is ready > https://github.com/oneapi-src/oneAPI-samples/pull/233

closed 2020-10-09 by xmnboy