oneapi-src/oneAPI-samples #243: Dependence on dpc_common.hpp

rscohn2, 2020-10-14:
The samples depend on dpc_common.hpp. It comes with the product dpc++ compiler, but is not part of the open source dpc++.

It would be better if the DPC++ samples would work with the open source DPC++, especially since the missing functionality is trivial.

anjgola, 2020-10-16:
Currently, this is by design. I am filing an internal feature request to track this.

JoeOster, 2020-11-18:
@xmnboy - any thoughts on this?

xmnboy, 2020-11-19:
@JoeOster -- when we worked with Paul Peterson on the solution to this problem (what works for the repo doesn't work for the sample browsers, and vice versa) we did not consider the open source version of the compiler.

There are few possible solutions, that I can see (could be others):

* Install the `dev-utilities` package and source the `dev-utilities/latest/env/vars.sh` script before compiling any samples you download or clone. This is a very light-weight package and can, on Linux, be easily installed via the `intel-oneapi-dev-utilities` package (e.g., `sudo apt install intel-oneapi-dev-utilities` on Ubuntu).

* Provide instructions on where to find the missing common include files. We already provide such a hint in the samples, but that hint only references them relative to the `dev-utilities` package. We could provide two references: one relative to the `dev-utilities` package (see above) and one relative to the samples repo (e.g., `-I /path/to/clone/folder/common`).

* Add symlinks to the common headers in every project to the relative location of the common folder in a clone. This is risky, because symlinks in a git repo don't always get reconstructed properly on Windows machines.

* Automatically copy the common headers folder into each sample in the repo so that a sample always includes those headers locally, within the sample project folder.

I prefer the first two options over the last two. @rscohn2, do you have any other ideas?

JoeOster, 2020-11-19:
I don't have an issue with using option 1&2  the other options propose larger risks

rscohn2, 2020-11-20:
You are already distribute the header files in /common so I added:
include_directories(../../../../common)
to the CMakeLists.txt.

JoeOster, 2021-02-08:
resolved for now

closed 2021-02-08 by JoeOster