oneapi-src/oneAPI-samples #249: fpga compiler command failed with exit code 1

Waxpple, 2020-10-15:
# Summary
fpga compiler command failed with exit code 1

# Steps to reproduce
By following the tutorial, I can't compile program for FPGA.
# Observed behavior
u50623@s001-n001:~/download/oneAPI-samples/DirectProgramming/DPC++/DenseLinearAlgebra/simple-add$ make hw -f Makefile.fpga
dpcpp -O2 -g -std=c++17 -fintelfpga a_buffers.o -o simple-add-buffers.fpga -Xshardware
aoc: Compiling for FPGA. This process may take several hours to complete.  Prior to performing this compile, be sure to check the reports to ensure the design will meet your performance targets.  If the reports indicate performance targets are not being met, code edits may be required.  Please refer to the oneAPI FPGA Optimization Guide for information on performance tuning applications for FPGAs.
Error (23035): Tcl error: 
Error (23031): Evaluation of Tcl script build/entry.tcl unsuccessful
Error: Quartus Prime Shell was unsuccessful. 2 errors, 0 warnings
For more detail, full Quartus compile output can be found in files quartuserr.tmp and quartus_sh_compile.log.
Error: Compiler Error, not able to generate hardware

dpcpp: error: fpga compiler command failed with exit code 1 (use -v to see invocation)
Makefile.fpga:33: recipe for target 'simple-add-buffers.fpga' failed
make: *** [simple-add-buffers.fpga] Error 1

What should I do?

akertesz, 2020-10-16:
Alby is a member of our FPGA customer support organization - she can help you or assign someone to do so.

AlbyB, 2020-10-16:
Hi there, it looks like you tried to execute the fpga compile on a DevCloud node that is not enabled to do so (s001-n001). Any FPGA compile jobs should be launched on nodes with the "fpga_compile" tag to ensure that you have the resources to complete it successfully. Any nodes not part of the "fpga_compile" group won't have the correct resources installed for an FPGA hardware compile. 
You can see the DevCloud Getting Started page to see an example of compiling for FPGA with vector add: https://devcloud.intel.com/oneapi/get-started/base-toolkit/#fpga-vector-add-sample-walkthrough

closed 2020-10-16 by AlbyB