oneapi-src/oneAPI-samples #314: OpenMP offloading options

jinz2014, 2020-11-04:
What are the OpenMP offloading options in Beta10 ?  Thanks.

:~/oneAPI-samples/DirectProgramming/C++/StructuredGrids/iso3dfd_omp_offload/build$ make
[ 25%] Building CXX object src/CMakeFiles/iso3dfd.dir/iso3dfd.cpp.o
**clang++: error: unsupported option '-fiopenmp'
clang++: error: unsupported option '-fopenmp-targets=spir64'**
src/CMakeFiles/iso3dfd.dir/build.make:62: recipe for target 'src/CMakeFiles/iso3dfd.dir/iso3dfd.cpp.o' failed
make[2]: *** [src/CMakeFiles/iso3dfd.dir/iso3dfd.cpp.o] Error 1
CMakeFiles/Makefile2:86: recipe for target 'src/CMakeFiles/iso3dfd.dir/all' failed
make[1]: *** [src/CMakeFiles/iso3dfd.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

$ which icpx
/opt/intel/oneapi/compiler/2021.1-beta10/linux/bin/icpx

JoeOster, 2020-11-04:
@slgogar - Can you provide some feedback on this request

slgogar, 2020-11-04:
> What are the OpenMP offloading options in Beta10 ? Thanks.
>
> :~/oneAPI-samples/DirectProgramming/C++/StructuredGrids/iso3dfd_omp_offload/build$ make
> [ 25%] Building CXX object [...]

Hi @jinz2014  ,

Can you please verify if your oneAPI HPC toolkit installation was successfully completed? Compiling openmp-offload application requires Intel(R) oneAPI HPC toolkit in addition to Intel(R) oneAPI base toolkit.

You can confirm openmp offload support as follows:

$ icpx --help | grep fopenmp-targets
 -->  **-fopenmp-targets=<value>**

Thanks,
Sunny

jinz2014, 2020-11-05:
Thanks for your answer.

closed 2020-11-05 by jinz2014