oneapi-src/oneAPI-samples #360: oneAPI for FPGA kernels may not be executing concurrently

kljucaric, 2020-11-18:
# Summary
Producer and Consumer kernels in the oneAPI pipe examples may not be executing concurrently. The reported start and end times for the kernels do not suggest concurrent execution. Is there a more appropriate way to collect overall FPGA profiling information than using the sycl::info commands inside of each kernel definition?

# Version
OneAPI Beta 10 (latest on Intel DevCloud)

# Environment
Intel DevCloud default

# Steps to reproduce
Followed steps listed at https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/DPC%2B%2BFPGA/Tutorials/Features/pipes using source code there.

# Observed behavior
Kernels are not executing concurrently. When adding profiling to the kernels, the first kernel (Producer) reports it start and end time, and the subsequent kernel's (Consumer) start time is after the first kernel's end time.

Profiling commands after q is submitted in the kernel definitions:
k_start = q.get_profiling_info<sycl::info::event_profiling::command_start>();
k_end = q.get_profiling_info<sycl::info::event_profiling::command_end>();

# Expected behavior
Consumer kernel's start time should be before Producer kernel's end time.

akertesz, 2020-11-23:
Do you see this in emulation or in FPGA hardware?

If it's in FPGA emulation, this behavior is completely expected. The FPGA emulator is nothing more than a CPU device that can handle FPGA extensions. It in no way reflects the performance (or concurrency) characteristics of the FPGA hardware. If it's in FPGA HW, that could only be an artifact of measurement or too-short kernel times. 

Either way, we'll look into it. Your confusion is legitimate and the tutorial ought to print this start/stop info and explain the results.

tyoungsc, 2020-11-23:
As Audrey said, I will be changing the tutorial to print this information and provide more details in the README. I would like to gently push about this point though:

> Consumer kernel's start time should be before Producer kernel's end time.

This is not always true. If the pipe depth is sufficiently high relative to the total amount of data, then the consumer kernel could start after the producer has finished. This tutorial sets a pipe depth of 4 and input size is 1024, so this shouldn't be the case in HW (if it is, it is likely a small granularity measurement issue). However, the emulator may use fixed size circular buffers that are larger than the requested depth to implement pipes (essentially ignoring the actual depth) and therefore you could be seeing the behaviour I just described in emulation.

akertesz, 2020-11-26:
Opened PR https://github.com/oneapi-src/oneAPI-samples/pull/380 to improve the pipes tutorial -- the improvements should prevent anyone else from being similarly mislead about the tool's behavior.

closed 2020-11-26 by akertesz

kljucaric, 2020-12-08:
Dropping in just for clarification.

The kernel performance profiled came from HW generated and executed designs. I assumed that the issue may have been an artifact of measurement as I wondered what is the best method for profiling in this situation. 

In any case, I appreciate the help and see the updated source.

akertesz, 2020-12-18:
@kljucaric  - Thanks for the follow-up and clarification. Interesting to know that you actually saw this in FPGA HW.

SYCL event profiling is the way to measure kernel start/stop times. (There's a neat [system profiling](https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/DPC%2B%2BFPGA/Tutorials/Tools/system_profiling) visualization tool you can use, but it uses the same mechanism under the hood.) While I'm not familiar with the details of exactly how the DPC++ runtime implements profiling under the hood, some measurement artifacts are to be expected. 

The problem with the (original) pipes tutorial was that the kernel run time was tiny with only 1024 values transacted. The actual duration of kernel execution was swamped by the runtime overhead, so the measurement artifacts swamped the signal. With 1,000,000 transactions, the results look as expected.

The updated tutorial allows you to control the input buffer size to get a feel for this.