oneapi-src/oneAPI-samples #385: Request to Add a oneAPI Sample OpenMP Offload Jupyter Notebook

karlqi2000, 2020-12-01:
# Summary
I have some OpenMP Offload tutorial jupyter notebooks and associated code that I would like to add to oneAPI samples (currently they are on the DevCloud). This is similar to the DPC++ essential training that's currently in the samples. 
I have both C++ and Fortran jupyter notebooks. The sample should reside in DirectProgramming/C++/Jupyter   and DirectProgramming/Fortran/Jupyter

# Problem statement
Having these notebooks and code samples would help introduce users to OpenMP Offload which is supported in the oneAPI HPC toolkit.

@pmpeter1 @JoeOster  @srdontha

JoeOster, 2020-12-02:
You are citing two domains, so long as the samples are unique to that domain, I have no real issues. Please supply the proposed folder names and paths

pmpeter1, 2020-12-02:
No objection.

closed 2021-01-11 by JoeOster