oneapi-src/oneAPI-samples #395: End-to-end-Workloads

cbigit, 2020-12-18:
# Summary
The examples under End-to-end-Workloads appear to be empty except for Census.   Seems like the contents for Morgage, NYTaxi, and Plastic have disappeared.

# Version
2021-1.gold

# Environment
As seen on github

# Steps to reproduce
https://github.com/oneapi-src/oneAPI-samples/tree/master/AI-and-Analytics/End-to-end-Workloads/Mortgage

# Observed behavior
empty directories: Mortgage, NYTaxi, Plastic only .gitkeep

# Expected behavior
Python examples etc

JoeOster, 2020-12-21:
Those folders are for future use and have no content as of yet. They should have been removed for the gold drop.

JoeOster, 2020-12-21:
PR384 will remove the folders

closed 2020-12-21 by JoeOster