oneapi-src/oneAPI-samples #417: Request to Add a oneAPI Sample fourier_correlation

hagabb, 2021-02-09:
## Summary
This is a request for a new code sample called fourier_correlation, which composes the Fourier correlation algorithm from oneMKL functions.

## Description
The Fourier correlation algorithm is commonly used to align 1D signals, overlay 2D images, perform 3D volumetric medical image registration, etc. The algorithm has high arithmetic intensity and the datasets are usually large so performance is critical.

The Fourier correlation algorithm is corr = IDFT(DFT(sig1) * CONJG(DFT(sig2))) where sig1 and sig2 are the real input data (e.g., 1D signals, 2D images, or 3D volumetric images), DFT is the discrete Fourier transform, IDFT is the inverse DFT, and CONJG is the complex conjugate. The necessary functions are available in oneMKL. In addition, oneMKL random number generators are used to add noise to the input data, which is a common technique in signal processing.

The two example codes implement the Fourier correlation algorithm using explicit buffering and USM. They show how to compose more complex mathematical operations from multiple functions while paying attention to where the data resides (to minimize host-device transfers) and where explicit synchronization is required vs. where synchronization is implicit in the task graph.

## Domain
Libraries

## Dependencies
None

## Proposed folder Location
oneAPI-samples/Libraries/oneMKL/fourier_correlation

JoeOster, 2021-02-10:
Approved by @srdontha and myself. Please open the PR and link it to this issue

hagabb, 2021-02-24:
I submitted a pull request. Please let me know if I did it correctly.

JoeOster, 2021-06-01:
PR#432

closed 2021-06-01 by JoeOster