oneapi-src/oneAPI-samples #444: Missing Fortran library after installation

Str-Gen, 2021-03-13:
Hi,

I have installed the latest version of intel-aikit (2021.1.1) on an Ubuntu 20.04 64-bit LTS.
Even the simplest use case with setting the environment variables once for a shell (as described in [https://software.intel.com/content/www/us/en/develop/documentation/get-started-with-ai-linux/top/before-you-begin.html#before-you-begin](https://software.intel.com/content/www/us/en/develop/documentation/get-started-with-ai-linux/top/before-you-begin.html#before-you-begin) does not lead to a functional environment.

Output of $ source setvars.sh

> :: initializing oneAPI environment ...
>    BASH version = 5.0.16(1)-release
> :: intelpython -- latest
> :: mkl -- latest
> :: iLiT -- latest
> :: modelzoo -- latest
> :: mpi -- latest
> :: ipp -- latest
> :: tbb -- latest
> :: dal -- latest
> :: dev-utilities -- latest
> :: compiler -- latest
> :: oneAPI environment initialized ::

Trying Sample Daal4py Linear Regression Example for Distributed Memory Systems [SPMD mode]
[https://github.com/oneapi-src/oneAPI-samples/tree/master/AI-and-Analytics/Getting-Started-Samples/IntelPython_daal4py_GettingStarted](https://github.com/oneapi-src/oneAPI-samples/tree/master/AI-and-Analytics/Getting-Started-Samples/IntelPython_daal4py_GettingStarted)

If I run the file directly through python: python IntelPython_daal4py_GettingStarted.py then I encounter a problem with the second import (the first line: import daal4py as d4p does finish).

> Traceback (most recent call last):
>   File "/tmp/IntelPython_daal4py_GettingStarted.py", line 36, in <module>
>     from sklearn.datasets import load_boston
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/__init__.py", line 80, in <module>
>     from .base import clone
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/base.py", line 21, in <module>
>     from .utils import _IS_32BIT
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/__init__.py", line 23, in <module>
>     from .class_weight import compute_class_weight, compute_sample_weight
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/class_weight.py", line 7, in <module>
>     from .validation import _deprecate_positional_args
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/validation.py", line 25, in <module>
>     from .fixes import _object_dtype_isnan, parse_version
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/fixes.py", line 18, in <module>
>     import scipy.stats
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/stats/__init__.py", line 388, in <module>
>     from .stats import *
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/stats/stats.py", line 174, in <module>
>     from scipy.spatial.distance import cdist
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/spatial/__init__.py", line 101, in <module>
>     from ._procrustes import procrustes
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/spatial/_procrustes.py", line 9, in <module>
>     from scipy.linalg import orthogonal_procrustes
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/linalg/__init__.py", line 194, in <module>
>     from .misc import *
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/linalg/misc.py", line 3, in <module>
>     from .blas import get_blas_funcs
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/linalg/blas.py", line 213, in <module>
>     from scipy.linalg import _fblas
> ImportError: libifport.so.5: cannot open shared object file: No such file or directory

I have tried to use the specific var setting shell file for the compiler, but that just messes up the variables so that it no longer sees intel_python. 

There are other shell files in the /opt/intel/oneapi folder named modulefiles-setup.sh and sys_check.sh.
 I have run modulefiles-setup.sh as root with output

> :: Initializing oneAPI modulefiles folder ...
> :: Removing any previous oneAPI modulefiles folder content.
> :: Generating oneAPI modulefiles folder links.
> -- compiler/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler
> -- compiler-rt/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler-rt
> -- compiler-rt32/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler-rt32
> -- compiler32/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler32
> -- compiler/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler
> -- compiler-rt/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler-rt
> -- compiler-rt32/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler-rt32
> -- compiler32/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler32
> -- dev-utilities/2021.1.1 -> /opt/intel/oneapi/dev-utilities/2021.1.1/modulefiles/dev-utilities
> -- dev-utilities/latest -> /opt/intel/oneapi/dev-utilities/latest/modulefiles/dev-utilities
> -- mkl/2021.1.1 -> /opt/intel/oneapi/mkl/2021.1.1/modulefiles/mkl
> -- mkl32/2021.1.1 -> /opt/intel/oneapi/mkl/2021.1.1/modulefiles/mkl32
> -- mkl/latest -> /opt/intel/oneapi/mkl/latest/modulefiles/mkl
> -- mkl32/latest -> /opt/intel/oneapi/mkl/latest/modulefiles/mkl32
> -- mpi/2021.1.1 -> /opt/intel/oneapi/mpi/2021.1.1/modulefiles/mpi
> -- mpi/latest -> /opt/intel/oneapi/mpi/latest/modulefiles/mpi
> -- tbb/2021.1.1 -> /opt/intel/oneapi/tbb/2021.1.1/modulefiles/tbb
> -- tbb/latest -> /opt/intel/oneapi/tbb/latest/modulefiles/tbb
> :: oneAPI modulefiles folder initialized.
> :: oneAPI modulefiles folder is here: "/opt/intel/oneapi/modulefiles"

Should the missing library be installed as part of installing intel-aikit or do I need another package to get this to work?

JoeOster, 2021-03-16:
@raoberman - can you take a look at this please

raoberman, 2021-03-22:
Hi @Str-Gen ,

I just ran the sample on Intel DevCloud and it ran successfully. A few questions on how you are running this sample: 

How did you install AI Kit (YUM, APT, offline, online, conda, etc.)? 

After running `source setvars.sh`, did you run command `source activate base` into the base environment (which includes all the packages you need to run this sample)? If not, try running `source activate base`  after running `source setvars.sh` and try running the sample again - this will put you in the base environment of the AI Kit.

You should not need to do an additional install outside of what is detailed in the README.md to run this sample. 

You can also look for more details and speedy assistance on the following Intel support forums:

[AI Analytics Toolkit Support Forum](https://community.intel.com/t5/Intel-oneAPI-AI-Analytics/bd-p/ai-analytics-toolkit)
[Intel Distribution for Python Support Forum](https://community.intel.com/t5/Intel-Distribution-for-Python/bd-p/distribution-python)
[oneDAL Support Forum](https://community.intel.com/t5/Intel-oneAPI-Data-Analytics/bd-p/oneapi-data-analytics-library)

Str-Gen, 2021-03-24:
Hi @raoberman,

I have just gone through a fresh install of intel-aikit on standard Ubuntu 20.04
The install happened via APT.
I have added the signing key and repo and after updating the package lists:
`apt install intel-aikit`
subsequently checked 

> /opt/intel/oneapi# ./modulefiles-setup.sh 
> :: Initializing oneAPI modulefiles folder ...
> :: Removing any previous oneAPI modulefiles folder content.
> :: Generating oneAPI modulefiles folder links.
> :: compiler/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler
> :: compiler-rt/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler-rt
> :: compiler-rt32/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler-rt32
> :: compiler32/2021.1.2 -> /opt/intel/oneapi/compiler/2021.1.2/modulefiles/compiler32
> :: compiler/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler
> :: compiler-rt/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler-rt
> :: compiler-rt32/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler-rt32
> :: compiler32/latest -> /opt/intel/oneapi/compiler/latest/modulefiles/compiler32
> :: dev-utilities/2021.1.1 -> /opt/intel/oneapi/dev-utilities/2021.1.1/modulefiles/dev-utilities
> :: dev-utilities/latest -> /opt/intel/oneapi/dev-utilities/latest/modulefiles/dev-utilities
> :: mkl/2021.1.1 -> /opt/intel/oneapi/mkl/2021.1.1/modulefiles/mkl
> :: mkl32/2021.1.1 -> /opt/intel/oneapi/mkl/2021.1.1/modulefiles/mkl32
> :: mkl/latest -> /opt/intel/oneapi/mkl/latest/modulefiles/mkl
> :: mkl32/latest -> /opt/intel/oneapi/mkl/latest/modulefiles/mkl32
> :: mpi/2021.1.1 -> /opt/intel/oneapi/mpi/2021.1.1/modulefiles/mpi
> :: mpi/latest -> /opt/intel/oneapi/mpi/latest/modulefiles/mpi
> :: tbb/2021.1.1 -> /opt/intel/oneapi/tbb/2021.1.1/modulefiles/tbb
> :: tbb/latest -> /opt/intel/oneapi/tbb/latest/modulefiles/tbb
> :: oneAPI modulefiles folder initialized.
> :: oneAPI modulefiles folder is here: "/opt/intel/oneapi/modulefiles"

and ran `. setvars.sh` 

> :: initializing oneAPI environment ...
>    -bash: BASH_VERSION = 5.0.17(1)-release
> :: compiler -- latest
> :: dal -- latest
> :: dev-utilities -- latest
> :: iLiT -- latest
> :: intelpython -- latest
> :: ipp -- latest
> :: mkl -- latest
> :: modelzoo -- latest
> :: mpi -- latest
> :: tbb -- latest
> :: oneAPI environment initialized ::

In the same session I run python interactive and get it to crash after just importing scikit-learn.
(Importing numpy and trying a simple operation works fine)

> Python 3.7.9 (default, Nov  4 2020, 21:30:07) 
> [GCC 7.3.0] :: Intel Corporation on linux
> Type "help", "copyright", "credits" or "license" for more information.
> Intel(R) Distribution for Python is brought to you by Intel Corporation.
> Please check out: https://software.intel.com/en-us/python-distribution
> import numpy as np
> a = np.random.random((1024,1024))
> b = np.random.random((1024,1024))
> np.dot(a,b)
> array([[260.5383183 , 251.93709464, 253.68565758, ..., 254.46630056,
>         250.27765743, 257.10592776],
>        [261.2331068 , 254.06810501, 253.56590212, ..., 253.48853209,
>         253.32285521, 252.86302982],
>        [257.73951127, 244.52420703, 248.93162864, ..., 251.88459298,
>         245.36563738, 253.4862064 ],
>        ...,
>        [247.54978111, 243.14576937, 243.90179437, ..., 252.25930381,
>         244.1008712 , 246.68352834],
>        [264.28225129, 254.0818712 , 253.95920525, ..., 262.71591147,
>         255.40575248, 256.15065389],
>        [262.73891822, 259.16666075, 257.10084824, ..., 265.55448841,
>         256.60828479, 262.18924677]])

> import sklearn as sk
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/__init__.py", line 80, in <module>
>     from .base import clone
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/base.py", line 21, in <module>
>     from .utils import _IS_32BIT
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/__init__.py", line 23, in <module>
>     from .class_weight import compute_class_weight, compute_sample_weight
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/class_weight.py", line 7, in <module>
>     from .validation import _deprecate_positional_args
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/validation.py", line 25, in <module>
>     from .fixes import _object_dtype_isnan, parse_version
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/sklearn/utils/fixes.py", line 18, in <module>
>     import scipy.stats
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/stats/__init__.py", line 388, in <module>
>     from .stats import *
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/stats/stats.py", line 174, in <module>
>     from scipy.spatial.distance import cdist
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/spatial/__init__.py", line 101, in <module>
>     from ._procrustes import procrustes
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/spatial/_procrustes.py", line 9, in <module>
>     from scipy.linalg import orthogonal_procrustes
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/linalg/__init__.py", line 194, in <module>
>     from .misc import *
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/linalg/misc.py", line 3, in <module>
>     from .blas import get_blas_funcs
>   File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/scipy/linalg/blas.py", line 213, in <module>
>     from scipy.linalg import _fblas
> ImportError: libifport.so.5: cannot open shared object file: No such file or directory

raoberman, 2021-03-26:
Hi @Str-Gen , 

Let me see if I can try to reproduce the issue. 

In the meantime, if you already have a Anaconda distribution installed, you can conda install AI Kit with daal4py, Intel optimized XGBoost and Modin via this install:

conda install -n aikit-modin intel-aikit-modin -c intel -c conda-forge

You can find more details, including a separate intel-daal4py install that does not include XGBoost or Modin, on the [AI Kit conda installation guide.](https://software.intel.com/content/www/us/en/develop/articles/installing-ai-kit-with-conda.html)

Let me know if this workaround works for you and if you are able to run the sample this way.

raoberman, 2021-05-21:
@Str-Gen I was unable to reproduce the issue. Are you still facing problems or should we close this issue?

raoberman, 2021-05-25:
This issue has been closed and we will no longer respond to this thread.  If you require additional assistance from Intel, please start a new thread.  Any further interaction in this thread will be considered community only.

JoeOster, 2021-05-25:
Closing due to lack of responses

closed 2021-05-25 by JoeOster