oneapi-src/oneAPI-samples #454: dpl_buffer.cpp seems seg-fault and output not consistent?

intelav, 2021-03-18:
# Summary
running the examples from oneapi-sample master led to inconsistent sorting output and then led to seg fault. 

# Version
One API Base tool Kit , ONEDPL policy execution 

# Environment
Device : Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz 
Ubuntu 18.04

# Steps to reproduce
Run examples in Jupyter/oneapi-essentials-training/07_DPCPP_Library/oneDPL_Introduction.ipynb and do the same in visual code (compile it using dpcpp -o <binary_name> dpl_buffer.cpp

# Observed behavior
first attempt sorted output is correct and second attempt led to seg fault
First Attempt ==> sorting a  array of integers in v{2,3,1,4} * 3 should have been 12 9 6 3
Device : Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
12 
9 
6 
3
Second Attempt ==>
Device : Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
./run_dpl_buffer.sh: line 6: 13425 Segmentation fault      (core dumped) ./dpl_buffer

Third Attempt on Visual code on same system, output is incorrect .. 
==========
Device : Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
0
3
0
6

Fourth Attempt on Visual Code
============
Device : Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
Segmentation fault (core dumped)
# Expected behavior
no seg fault and output should be consistent and correct ..

intelav, 2021-03-18:
[image]

praveenkk123, 2021-03-24:
Hi, Thanks for notifying us about the issue. Just wanted to let you know that the samples in Jupyter/oneapi-essentials-training are tested on Intel Devcloud and works good on the Jupyter environment on Devcloud. We are planning to test these using different environments and different IDEs soon and we will keep a note of this issue. Until we test and release the Jupyter/oneapi-essentials-training for other IDEs please run  the samples on Intel Devcloud for better experience.

 Below are the steps to access these Jupyter notebooks on Intel Devcloud

Register on Intel Devcloud : https://intelsoftwaresites.secure.force.com/devcloud/oneapi

Go to the "Terminal" in the Intel Devcloud
Type in the below command to download the oneAPI-essentials series notebooks into your Devcloud account /data/oneapi_workshop/get_jupyter_notebooks.sh
Navigate to oneAPI_Essentials folder and open the Welcome.ipynb

Thanks
Praveen

JoeOster, 2021-04-09:
@intelav - I plan to close this issue in next 24 hrs unless further action is required

intelav, 2021-04-10:
Even without Jupyter environment , I noticed the issue.. on Ubuntu 18.04

JoeOster, 2021-06-01:
@praveenkk123 - can you take a look at this please

JoeOster, 2021-06-03:
We are unable to replicate what @intelav is seeing. Probably some environmental issues on his machine. 
We are checking with oneDPL library team for some additional guidance

avaish1, 2021-06-17:
Sorry for not following on this issue for long. You can close this issue if not an issue with Dev Cloud. I just happened to test on an ubuntu 18.04 m/c and notified here. 

Thanx for all your Effort!!
intelav

JoeOster, 2021-06-17:
Thank you for the feedback

closed 2021-06-17 by JoeOster