oneapi-src/oneAPI-samples #490: Offload time of iso2dfd is about 100 times slower than the CPU time

aljohnston112, 2021-05-01:
Here is the output:

`
Initializing ...
Grid Sizes: 1000 1000
Iterations: 2000

Computing wavefield in device ..
 Running on       Intel(R) Core(TM) i7-2760QM CPU @ 2.40GHz
 The Device Max Work Group Size is : 8192
 The Device Max EUCount is : 8
Offload time: 3186.34 s

Computing wavefield in CPU ..
Initializing ...
CPU time: 39.8897 s

Final wavefields from device and CPU are equivalent: Success
Final wavefields (from device and CPU) written to disk
Finished.
`

I am assuming both simulations are run on the CPU because my GPU is unsupported, so what is making it run so much slower?

JoeOster, 2021-05-04:
@Alberto-Villarreal - can you look into this issue please

JoeOster, 2021-05-20:
@Alberto-Villarreal - can you look into this issue please.

Alberto-Villarreal, 2021-05-29:
@aljohnston112 Offloading to the CPU might be slower, because of some overhead, but I would not expect that slow. I'm not able to replicate. Could you please contact me by email directly to investigate a possible issue in your system?

JoeOster, 2021-06-01:
Ticket will be worked offline, closing for now

closed 2021-06-01 by JoeOster