oneapi-src/oneAPI-samples #495: Request to Add a oneAPI Sample OpenCLInterop

karlqi2000, 2021-05-04:
## Summary
This is a request for a new code Sample called `DPC++ OpenCL Interoperabilty`

## Purpose
These code sample will show how OpenCL objects and Kernels can interact with DPC++. This is useful for OpenCL programmers who wants to migrate to DPC++ in a piecemeal manner, making it easier to migrate from OpenCL.

## Domain
Direct Programming / DPC++

## Description
Two examples here, one showing DPC++ compiling and launching an OpenCL Kernel. Another showing how DPC++ can use OpenCL memory, context, platform, device, kernel objects.

## Proposed folder Location
oneAPI-samples/DirectProgramming/DPC++/OpenCLInterop

@JoeOster  @pmpeter1 @bjodom @moushumi-maria

JoeOster, 2021-05-04:
Sample proposal ppt has been submitted to the WG, with the proposal being on 05/07/21

JoeOster, 2021-05-07:
Approved by working group

closed 2021-06-01 by JoeOster