oneapi-src/oneAPI-samples #510: zero copy example does not compile on devcloud

Hecmay, 2021-05-17:
# Summary
I am trying run the zero-copy example on devcloud. It seems that the default platform specified in the CMakelist.txt cannot be found on devcloud, so I changed it to `pac_s10_dc`. And I got the compilation error from aoc compiler as detailed below.  

Is this example supported on devcloud S10 board? If not, where can I find a board that supports USM and zero-copy? 

# Version
I am on the latest commit of master branch. Here is the dpcpp version 
```shell
u68165@s001-n143:~$ dpcpp -v
Intel(R) oneAPI DPC++ Compiler 2021.2.0 (2021.2.0.20210317)
```
For Quartus and BSP version

```shell
u68165@s001-n143:~$ tools_setup -t S10DS
sourcing /glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/init_env.sh
export QUARTUS_HOME=/glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/quartus
export OPAE_PLATFORM_ROOT=/glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/d5005_ias_2_0_1_b237
export AOCL_BOARD_PACKAGE_ROOT=/glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/d5005_ias_2_0_1_b237/opencl/opencl_bsp
Adding $OPAE_PLATFORM_ROOT/bin to PATH
export INTELFPGAOCLSDKROOT=/glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/hld
export ALTERAOCLSDKROOT=/glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/hld
Adding $QUARTUS_HOME/bin to PATH
source /glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/hld/init_opencl.sh
```

# Environment
Intel devcloud cluster. S10 node with oneAPI
```shell
--------------------------------------------------------------------------------------
Nodes with Stratix 10 OneAPI: (1 available/3 total)
s001-n143
--------------------------------------------------------------------------------------
```

# Steps to reproduce
Just clone the repo, generate Makefile and run `make fpga` as mentioned in the instructions.

# Observed behavior
Here is what I got after running `make fpga`

```shell
u68165@s001-n144:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/DesignPatterns/zero_copy_data_transfer/build$ make fpgaScanning dependencies of target zero_copy_data_transfer.fpga
[ 50%] Building CXX object src/CMakeFiles/zero_copy_data_transfer.fpga.dir/zero_copy_data_transfer.cpp.o[100%] Linking CXX executable ../zero_copy_data_transfer.fpga
aoc: Running OpenCL parser....
Error: SPIRV to LLVM IR FAILED
dpcpp: error: fpga compiler command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/zero_copy_data_transfer.fpga.dir/build.make:94: recipe for target 'zero_copy_data_transfer.fpga' failedmake[3]: *** [zero_copy_data_transfer.fpga] Error 1CMakeFiles/Makefile2:229: recipe for target 'src/CMakeFiles/zero_copy_data_transfer.fpga.dir/all' failed
make[2]: *** [src/CMakeFiles/zero_copy_data_transfer.fpga.dir/all] Error 2
CMakeFiles/Makefile2:268: recipe for target 'src/CMakeFiles/fpga.dir/rule' failed
make[1]: *** [src/CMakeFiles/fpga.dir/rule] Error 2
Makefile:183: recipe for target 'fpga' failed
make: *** [fpga] Error 2
```

# Expected behavior
to generate bitstream successfully.

JoeOster, 2021-05-20:
@tyoungsc - Can you or someone on your team take a look at this please

mdbtucker, 2021-05-21:
I believe this error has been mis-characterized.  I just tried running cmake then make fpga on the devcloud, and this is the error I got:
----------------------
u73448@login-2:~/samples.1/DirectProgramming/DPC++FPGA/Tutorials/DesignPatterns/zero_copy_data_transfer/build$ make fpga
Scanning dependencies of target zero_copy_data_transfer.fpga
[ 50%] Building CXX object src/CMakeFiles/zero_copy_data_transfer.fpga.dir/zero_copy_data_transfer.cpp.o
[100%] Linking CXX executable ../zero_copy_data_transfer.fpga
warning: -reuse-exe file '/home/u73448/samples.1/DirectProgramming/DPC++FPGA/Tutorials/DesignPatterns/zero_copy_data_transfer/build/zero_copy_data_transfer.fpga' not found; ignored
Out of memory in module quartus_sh (1023 megabytes used).
Error: The patches required to compile for the target board (0.05dcp) is not installed for the following Quartus:
/glob/development-tools/versions/oneapi/2021.2/inteloneapi/intelfpgadpcpp/2021.2.0/QuartusPrimePro/19.2/quartus/bin/quartus_sh

dpcpp: error: fpga compiler command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/zero_copy_data_transfer.fpga.dir/build.make:94: recipe for target 'zero_copy_data_transfer.fpga' failed
make[3]: *** [zero_copy_data_transfer.fpga] Error 1
CMakeFiles/Makefile2:229: recipe for target 'src/CMakeFiles/zero_copy_data_transfer.fpga.dir/all' failed
make[2]: *** [src/CMakeFiles/zero_copy_data_transfer.fpga.dir/all] Error 2
CMakeFiles/Makefile2:268: recipe for target 'src/CMakeFiles/fpga.dir/rule' failed
make[1]: *** [src/CMakeFiles/fpga.dir/rule] Error 2
Makefile:183: recipe for target 'fpga' failed
make: *** [fpga] Error 2
-------------------

So the platform used in this sampe (intel_s10sx_pac:pac_s10_usm) IS found, but the Quartus version installed on the DevCloud does not include the required patch.  This is not a problem with the sample, it is a problem with the binaries installed on the DevCloud.

@JoeOster or @Hecmay can you help me figure out who this issue should be re-directed to?

mdbtucker, 2021-05-21:
Alby, I'm assigning this to you as it seems to be a problem with the FPGA Add-on installed on the Dev Cloud.  It's not clear to me if the wrong add on is installed, or if there's a problem with the add-on that was shipped and it doesn't support the USM board.

AlbyB, 2021-05-21:
Hi there, I think the root of the error is a misunderstanding about what DevCloud nodes to use. It looks like you're using n144, which is part of the fpga_runtime:stratix10 group, meant for running the designs after they've been compiled. However, the runtime nodes don't necessarily have Quartus installed. If you want to compile your design for hardware, you should be using the fpga_compile nodes to submit your jobs, ex: 

qsub -l nodes=1:fpga_compile:ppn=2 -d . build_fpga_hw.sh

These don't have a board, but do have Quartus. Running it on one of these nodes (ex: n056) goes through the aoc stage fine for me, so please let me know if you still have problems. You can see more about the flow/job submission here: 
https://devcloud.intel.com/oneapi/documentation/base-toolkit/
https://devcloud.intel.com/oneapi/documentation/job-submission/

mdbtucker, 2021-05-21:
Thanks Alby, I made a similar mistake (I used the login node to try to run my FPGA compile).  I've launched a compile on an fpga_compile node, and while it's still in progress it did not error out immediately as I saw previously, so I assume this will be fine.  Sorry for not catching this myself, I don't have much experience using DevCloud (obviously :-) ).

closed 2021-05-21 by mdbtucker

Hecmay, 2021-05-21:
Thanks @AlbyB @mdbtucker. At the beginning I was trying to use the fpga_compile node, but the qsub request hanged forever. So I switched to FPGA node instead. I will give it a try later.

AlbyB, 2021-05-21:
If they're all still busy and you know exactly what you want to execute, you can also try writing it up in a .sh file and submitting that to the fpga_compile group, and it'll run whenever there's a free node.

Hecmay, 2021-05-21:
It is being compiled now. Thanks!

One thing to notice is that: do not run `tools_setup -t S10DS` as suggested by the devcloud document before the compilation. This will mess things up somehow.

```shell
u68165@s001-n053:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/DesignPatterns/zero_copy_data_transfer/
build$ make fpga
Scanning dependencies of target zero_copy_data_transfer.fpga
[ 50%] Building CXX object src/CMakeFiles/zero_copy_data_transfer.fpga.dir/zero_copy_data_transfer.cpp.o
[100%] Linking CXX executable ../zero_copy_data_transfer.fpga
Board 'pac_s10' not found.
Searched in the board package at:
  /glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/d5005_ias_2_0_1_b237/opencl/open$
l_bsp
Board list:
  pac_s10_dc
     Board Package: /glob/development-tools/versions/fpgasupportstack/d5005/2.0.1/inteldevstack/d5005_ias_2_0_$
_b237/opencl/opencl_bsp

If you are using a 3rd party board, please ensure:
  1) The board package is installed (contact your 3rd party vendor)
  2) You have used -board-package=<bsp-path> to specify the path to
     your board package installation
Error: No board_spec.xml found for board 'pac_s10' (Searched for: /glob/development-tools/versions/fpgasupport$
tack/d5005/2.0.1/inteldevstack/d5005_ias_2_0_1_b237/opencl/opencl_bsp/hardware/pac_s10/board_spec.xml).
dpcpp: error: fpga compiler command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/zero_copy_data_transfer.fpga.dir/build.make:94: recipe for target 'zero_copy_data_transfer.fpga$
 failed
make[3]: *** [zero_copy_data_transfer.fpga] Error 1
CMakeFiles/Makefile2:229: recipe for target 'src/CMakeFiles/zero_copy_data_transfer.fpga.dir/all' failed
make[2]: *** [src/CMakeFiles/zero_copy_data_transfer.fpga.dir/all] Error 2
```

AlbyB, 2021-05-21:
No problem, glad to hear that it's working!

If you wouldn't mind, could you link what documentation you're following and let us know how you came across it/these samples?

Hecmay, 2021-05-22:
@AlbyB Here is the document I referred to: https://github.com/intel/FPGA-Devcloud/tree/master/main/Devcloud_Access_Instructions#53-development-tool-access-and-setup. I asked Larry before (he is the maintainer of devcloud cluster I believe?), and he forwarded me this link.

Unfortunately, my interactive job on fpga_compile node was terminated because of ssh broken pipe. And my batch job to fpag_compile node is hanging forever again (even though I can see many available compilation-only nodes in the cluster). I guess I need to try my luck on fpga_runtime node again. lol

Hecmay, 2021-05-22:
@AlbyB 

Sorry to bother you again.. The compilation on compile_fpga can run successfully and it can generate `.fpga` file as expected. However, the bitstream cannot be executed on real FPGA on devcloud.

Here is the detailed error message
```shell
u68165@s001-n143:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/DesignPatterns/zero_copy_data_transfer/build$ ./zero_copy_data_transfer.fpga 
Error enumerating AFCs: not found
Error enumerating AFCs: not found
Error enumerating AFCs: not found
Error enumerating AFCs: not found
Segmentation fault
```