oneapi-src/oneAPI-samples #513: Error in the execution of PyTorch_Hello_World.py

Andrea8Testa, 2021-05-21:
Following the tutorial to make this sample work, after the installation of the oneAPI AI Analytics Toolkit, I executed the commands:
```

. /opt/intel/oneapi/setvars.sh
conda activate pytorch
cd /opt/intel/oneapi/intelpython/latest/envs/pytorch (where I cloned the script, as administrator)
python PyTorch_Hello_World.py 
```

I get the error:

`Segmentation fault (core dumped)`

Instead, if I try to clone the environment:

`conda create --name usr_pytorch --clone pytorch`

the error is:

```
Source:      /opt/intel/oneapi/intelpython/latest/envs/pytorch
Destination: /home/franka/.conda/envs/usr_pytorch
The following packages cannot be cloned out of the root environment:
 - file:///opt/intel/oneapi/conda_channel/linux-64::conda-4.9.2-py37hea4d9f2_0
Packages: 74
Files: 204

Downloading and Extracting Packages
cpuonly-1.0          | ########################################################################################################################################################## | 100% 
dataclasses-0.8      | ########################################################################################################################################################## | 100% 
python_abi-3.7       | ########################################################################################################################################################## | 100% 

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/exceptions.py", line 1079, in __call__
        return func(*args, **kwargs)
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/cli/main.py", line 84, in _main
        exit_code = do_call(args, p)
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/cli/conda_argparse.py", line 83, in do_call
        return getattr(module, func_name)(args, parser)
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/cli/main_create.py", line 41, in execute
        install(args, parser, 'create')
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/cli/install.py", line 222, in install
        clone(args.clone, prefix, json=context.json, quiet=context.quiet, index_args=index_args)
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/cli/install.py", line 74, in clone
        index_args=index_args)
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/misc.py", line 290, in clone_env
        force_extract=False, index_args=index_args)
      File "/opt/intel/oneapi/intelpython/latest/lib/python3.7/site-packages/conda/misc.py", line 90, in explicit
        assert not any(spec_pcrec[1] is None for spec_pcrec in specs_pcrecs)
    AssertionError

`$ /opt/intel/oneapi/intelpython/latest/bin/conda create --name usr_pytorch --clone pytorch`

  environment variables:
                 CIO_TEST=<not set>
                CLASSPATH=/opt/intel/oneapi/mpi/2021.2.0//lib/mpi.jar:/opt/intel/oneapi/dal/2021
                          .2.0/lib/onedal.jar
        CMAKE_PREFIX_PATH=/opt/intel/oneapi/tbb/2021.2.0/env/..:/opt/intel/oneapi/dal/2021.2.0:/
                          home/franka/elia_ws/devel:/home/franka/catkin_ws/devel:/home/franka/ws
                          _moveit/devel:/opt/ros/melodic
                CONDA_EXE=/opt/intel/oneapi/intelpython/latest/bin/conda
         CONDA_PYTHON_EXE=/opt/intel/oneapi/intelpython/latest/bin/python
               CONDA_ROOT=/opt/intel/oneapi/intelpython/latest
              CONDA_SHLVL=0
                    CPATH=/opt/intel/oneapi/tbb/2021.2.0/env/../include:/opt/intel/oneapi/mpi/20
                          21.2.0//include:/opt/intel/oneapi/mkl/latest/include:/opt/intel/oneapi
                          /ipp/2021.2.0/include:/opt/intel/oneapi/dev-utilities/2021.2.0/include
                          :/opt/intel/oneapi/dal/2021.2.0/include:/opt/intel/oneapi/compiler/202
[... 97 lines cut ...]
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /opt/intel/oneapi/intelpython/latest  (read only)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          file:///opt/intel/oneapi/conda_channel/linux-64
                          file:///opt/intel/oneapi/conda_channel/noarch
                          https://conda.anaconda.org/intel/linux-64
                          https://conda.anaconda.org/intel/noarch
          package cache : /opt/intel/oneapi/intelpython/latest/pkgs
                          /home/franka/.conda/pkgs
       envs directories : /home/franka/.conda/envs
                          /opt/intel/oneapi/intelpython/latest/envs
               platform : linux-64
             user-agent : conda/4.9.2 requests/2.25.1 CPython/3.7.9 Linux/5.6.19-rt11 ubuntu/18.04.5 glibc/2.27
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.
```

How can be solved?

JoeOster, 2021-05-27:
@ louie-tsai Can you look at this and work with the customer on it.

JoeOster, 2021-06-01:
@jingxu10 - Can you take a look at this please

jingxu10, 2021-06-02:
Which version have you tried? How did you installed it? I just tried on ai kit docker image (https://hub.docker.com/r/intel/oneapi-aikit), the hello world script worked correctly.

Andrea8Testa, 2021-06-02:
I tried the Intel® oneAPI AI Analytics Toolkit (https://software.intel.com/content/www/us/en/develop/tools/oneapi/ai-analytics-toolkit.html#gs.2chtw8) for Linux operating system selecting the Web & Local distribution. I chose the online installer and I used the two commands shown in the Command Line Installation section. After the installation I followed this guide (https://software.intel.com/content/www/us/en/develop/documentation/get-started-with-ai-linux/top/before-you-begin.html#before-you-begin) for the set up. All the scripts are executed without errors, so I thought that the installation was executed correctly.
The hello world script seems not to be included in this toolkit, so I had to clone it from this Github Repository. I tried to clone it inside and outside the environment "pytorch" but the result remained the same.

jingxu10, 2021-06-03:
I cannot reproduce the error you reported.
The code ran successfully with the following steps on my machine.
1. wget https://registrationcenter-download.intel.com/akdlm/irc_nas/17766/l_AIKit_p_2021.2.0.1101_offline.sh
2. sudo bash l_AIKit_p_2021.2.0.1101_offline.sh
3. source /opt/intel/oneapi/setvars.sh
4. conda activate pytorch
5. wget https://raw.githubusercontent.com/oneapi-src/oneAPI-samples/master/AI-and-Analytics/Getting-Started-Samples/IntelPyTorch_GettingStarted/PyTorch_Hello_World.py
6. python PyTorch_Hello_World.py

Andrea8Testa, 2021-06-04:
Ok thank you, I tried your steps and it actually works. Maybe I did something wrong when I followed the get started guide (https://software.intel.com/content/www/us/en/develop/documentation/get-started-with-ai-linux/top.html). 

Anyway I am still not able to run this code on the GPU, that is the real reason why I downloaded this toolkit (since I have an Intel graphic card). 

If I add to the code the lines:
```
    device = "cuda" if torch.cuda.is_available() else "cpu"
    print("Using {} device".format(device))
```
It still not able to find any GPU.

jingxu10, 2021-06-04:
Intel GPU support has not been enabled in public version of oneAPI package yet. The packages in current oneAPI AI Analytics Toolkit are all for CPU for now. We will release the packages later once they are ready to go.

Andrea8Testa, 2021-06-04:
Ok, so it seems there are no options to use GPU with Intel graphic card.
Anyway, thank you very much for the help.

closed 2021-06-06 by Andrea8Testa