oneapi-src/oneAPI-samples #585: Initialize environment variables gives error

masoudshao, 2021-07-12:
# Summary
Running 

     >Intel oneAPI: Initialize environment variables

in VS code in Mac return Error 1 
[image]

# Version
oneAPI Base Toolkit version 2021.2
oneAPI HPC Toolkit version 2021.3

# Environment
Operating system:
MacOS Catalina 10.15.7

Visual Studio COde:
Version: 1.58.0 (Universal)
Commit: 2d23c42a936db1c7b3b06f918cde29561cc47cd6
Date: 2021-07-08T06:54:17.694Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Darwin x64 19.6.0

# Steps to reproduce
I haven't cloned the repo so not sure if it is reproduceable with master branch

# Observed behavior
Another notification pops up after the error that says all new terminals will have their environment set. Also, another pop-up window on the bottom right of VS appears that says it found the setvars.sh script. However, opening a new terminal in VS and typing "ifort --version" returns error. 

# Expected behavior
The result from the get started tutorials

tingleby, 2021-07-15:
Perhaps related to https://github.com/intel/vscode-environment-variables/issues/5

JoeOster, 2021-07-15:
@masoudshao - this seems to be an issue with another project. Please enter an issue on the [VSCode environmental Variable Project](https://github.com/intel/vscode-environment-variable). 

I have reported this to them, but an issue ticket from you is recommended.

closed 2021-07-15 by JoeOster