oneapi-src/oneAPI-samples #654: GPU training

jackyko1991, 2021-09-06:
Is the oneAPI possible to train AI with gpu?

Current pytorch and tensorflow examples are calling mkldnn. With oneDNN it claims that can run on opencl (AMD), intel xe and nvidia gpu, any example to show that it is running on cpu or gpu devices?

JoeOster, 2021-09-21:
@louie-tsai or @raoberman - Can either of you address this question please, as it is AI related

louie-tsai, 2022-07-23:
@jackyko1991 
Sorry for late response.
The GPU support will be released when Intel Ponte Vecchio dGPU is launched.

For early access, you might refer to below page once you have a public DevCloud account, and check for the support section.
https://devcloud.intel.com/oneapi/early_access_resources/
[image]

Let us know if you already have a  early Intel Ponte Vecchio dGPU sample on hand, and we could help on related software installation.

troy818, 2023-04-25:
OneSmpl_Team1 working on this

troy818, 2023-04-25:
(OneSmpl_Team1)

@jackyko1991 

The GPU support for pyTorch training has been released as intel_extension_for_pytorch, check the links for your reference.
https://intel.github.io/intel-extension-for-pytorch/xpu/latest/index.html
https://github.com/intel/intel-extension-for-pytorch

Please note the Verified Hardware Platforms:
Intel® Data Center GPU Flex Series 170
Intel® Data Center GPU Max Series
Intel® Arc™ A-Series GPUs (Experimental support)

jimmytwei, 2023-08-12:
@jackyko1991 Closing this issue. As @troy818 mentioned above, we have support for dGPUs.

closed 2023-08-12 by jimmytwei