oneapi-src/oneAPI-samples #660: Default optimization

chuckyount, 2021-09-10:
The README states that the default build is the baseline w/o any optimizations. It appears that the default build is actually level-3 optimization. To build without optimization, you have to use `cmake -DUSE_OPT3=0 ..`.

chuckyount, 2021-09-10:
Referring to this README: https://github.com/oneapi-src/oneAPI-samples/blob/master/DirectProgramming/C%2B%2B/StructuredGrids/iso3dfd_omp_offload/README.md

JoeOster, 2021-09-21:
@slgogar - Hey Sunny, can you loo into this please. If possible a fix asap would be appreciated as we are in validation now

slgogar, 2021-09-22:
@JoeOster @chuckyount  i am not able to reproduce the problem mentioned above.

[slgogar@hostname  iso3dfd_omp_offload]$ git diff src/CMakeLists.txt
[slgogar@hostname iso3dfd_omp_offload]$ git diff CMakeLists.txt
[slgogar@hostname  iso3dfd_omp_offload]$ cd build_test/
[slgogar@hostname  build_test]$ source /opt/intel/oneapi/setvars.sh

<snip>

[slgogar@hostname build_test]$ cmake ..
-- The C compiler identification is GNU 8.4.1
-- The CXX compiler identification is Clang 13.0.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/intel/oneapi/compiler/2021.3.0/linux/bin/icpx
-- Check for working CXX compiler: /opt/intel/oneapi/compiler/2021.3.0/linux/bin/icpx -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
**-- Using Baseline target code**
-- Configuring done
-- Generating done
-- Build files have been written to: <workdir>/oneAPI-samples/DirectProgramming/C++/StructuredGrids/iso3dfd_omp_offload/build_test

If you already compiled for some optimization path for example -  "**cmake -DUSE_OPT1=1 .**."  then in order to go back to the baseline version you will have to either clean the build directory (**rm -rf build**) or use "**cmake -DUSE_OPT1=0 ..**" OR use "**ccmake ..**" to configure the optimization level as desired. Chosen optimized version is printed in cmake output for users reference.

chuckyount, 2021-10-04:
Yes, the problem was as you mentioned: I had used OPT1, then tried to go back to baseline. Maybe this scenario should be better documented.

JoeOster, 2021-10-08:
@slgogar - Hey What are your thoughts on documentation improvements?

bopeng1234, 2023-04-25:
OneSmpl_Team1 working on this

commit 73096e27 (2023-04-27): Merge pull request #1581 from bopeng1234/OMP_iso3dfd_readme_issue_660

C++, iso3dfd, update README.md, fix #660
commit e4c0271d (2023-04-27): C++, iso3dfd, update README.md, fix #660