oneapi-src/oneAPI-samples #670: Cannot override optimization level set for particle-diffusion sample

serge-intel, 2021-09-16:
# Summary
particle-diffusion/src/CMakeLists.txt sets default optimization level to -O3, regardless of the CMAKE_BUILD_TYPE

# Version
oneAPI Base Tookit 2021.4

# Environment
Ubuntu 20.04
Coffee Lake / GEN9 graphics

# Steps to reproduce
from particle-diffusion sample directory run:
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=DEBUG ..

# Observed behavior
Running "cmake -DCMAKE_BUILD_TYPE=DEBUG .." results in both -O0 and -O3 specified in flags.make:
CMakeFiles/motionsim.exe.dir/flags.make:CXX_FLAGS = -g -O0 -O3 -std=c++17 -g
As the result, dpcpp seem to performs -O3 optimization, defeating the purpose of DEBUG build

# Expected behavior
cmake uses -O0 -g compiler flags when CMAKE_BUILD_TYPE=DEBUG is set

JoeOster, 2021-09-21:
@Alberto-Villarreal -can you loo into this please

troy818, 2023-04-25:
OneSmpl_Team1 working on this

troy818, 2023-04-25:
(OneSmpl_Team1)

Now only use O3 for cmake.