oneapi-src/oneAPI-samples #750: Possible logic error in dpc_reduce example code

hagabb, 2021-11-16:
I think the following code has a logic error in lines 657-666: https://github.com/oneapi-src/oneAPI-samples/blob/master/DirectProgramming/DPC%2B%2B/ParallelPatterns/dpc_reduce/src/main.cpp. 

[image]

The if-test on line 657 has MPI rank-0 calling the DPC++ kernels to force JIT compilation. The programmer assumes that the rest of the MPI communicator will have access to the master’s cached JIT compilations. The other MPI ranks are separate processes, so they won’t have access. Contrary to the programmer’s expectations (specified on lines 660-662), the other MPI ranks will still perform the JIT compilations when the DPC++ kernels are first invoked.

lqnguyen, 2022-01-14:
Hi Henry,

Thank you for pointing out the concern. But in the application, only the master MPI rank (rank 0) needs to execute these kernels (lines 663, 664, 665, 666), and other MPI ranks don't. Therefore, the other ranks don't need to have access to master's cache JIT compilation.

The three comment lines (660, 661, 662) are not correct and should be removed.

hagabb, 2022-01-14:
Thanks Loc. I'll update the code in my fork and submit a pull request.

closed 2022-01-14 by hagabb