oneapi-src/oneAPI-samples #754: tbb library linking issue

pravin25, 2021-11-18:
# Summary
Trying to compile oneapi-sample code example  dpc_reducer( https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/DPC%2B%2B/ParallelPatterns/dpc_reduce)

# Version
oneAPI Toolkit version: l_BaseKit_p_2021.4.0.3422_offline.sh
oneAPI Sample version:  latest master branch.

# Environment
OS: Ubuntu 20.04
Hardware : NUC9

# Steps to reproduce
followed dpc_reduce example README file.
export I_MPI_CXX=dpcpp
cmake ../
make 

# Observed behavior
following compile time error is coming.
_**_/usr/bin/ld: /tmp/main-585a82.o: in function `task_group_context':
/opt/intel/oneapi/tbb/2021.4.0/env/../include/tbb/../oneapi/tbb/task_group.h:271: undefined reference to `tbb::detail::r1::initialize(tbb::detail::d1::task_group_context&)'_**_

# Expected behavior
should be compile successfully.

pravin25, 2021-11-19:
this issue is coming due to libtbb.
if in Ubuntu system already libtbb is installed then oneapi environment variable is set that library path instead of oneapi libtbb library path.

sau0, 2023-04-26:
OneSmpl_Team3 working on this.