oneapi-src/oneAPI-samples #755: Failure in executing openmp reduction sample on GPU when using Tigerlake platform

saratpoluri, 2021-11-18:
# Summary
Provide a short summary of the issue. Sections below provide guidance on what
factors are considered important to reproduce an issue.
Build and run openmp reduction. I see the below error when it tries to run the device version on GPU:
Libomptarget --> Unable to generate entries table for device id 0.
Libomptarget --> Failed to init globals on device 0
Libomptarget error: Source location information not present. Compile with -g or -gline-tables-only.
Libomptarget fatal error 1: failure of target construct while offloading is mandatory

# Version
Report oneAPI Toolkit version and oneAPI Sample version or hash.
Oneapi samples git commit: 4bed52e76ceb17243a0bc4ce24e9aed52aaa6e49

# Environment
Provide OS information and hardware information if applicable.
Ubuntu 20.04
11th Gen Intel(R) Core(TM) i7-1185GRE @ 2.80GHz

# Steps to reproduce
Please check that the issue is reproducible with the latest revision on
master. Include all the steps to reproduce the issue. 
build openmp reduction sample and do make run.

# Observed behavior
Document behavior you observe. For performance defects, like performance
regressions or a function being slow, provide a log if possible.
Ran the sample with LIBOMPTARGET_DEBUG=2 and observed the following:
error: double type is not supported on this platform
in kernel: 'openmp_device_calc_pi(int)'
error: backend compiler failed build.

# Expected behavior
The reduction sample runs on GPU and delivers the result of pi.

#Verified Fix:
I don't have permissions to check code in. The fix is to add a conversion to float for the denominator on line 63:
sum = sum + 4.0f / (**float**)(1.0 + x * x);

saratpoluri, 2021-11-18:
Similar issues exist in other samples as well. Are the samples tested on TGL iGPU with the latest OneAPI release (2021.4)?

bopeng1234, 2023-04-25:
OneSmpl_Team1 working on this

commit 5f746c1e (2023-04-27): Fix OMP Reduction sample, on TGL GPU kernel use float, fix #755
commit ff69355c (2023-04-27): Merge pull request #1580 from bopeng1234/fix_omp_reduction_on_TGL_issue_755

Fix OMP Reduction sample, on TGL GPU kernel use float, fix #755
closed 2023-07-13 by jimmytwei