oneapi-src/oneAPI-samples #887: Initiation interval links are broken

paul-white-intel, 2022-03-11:
# Summary
The links to the documentation (e.g. Initiation Interval, loop-carried dependencies, -Xsclock) are broken.

# URLs
https://github.com/oneapi-src/oneAPI-samples/blob/master/DirectProgramming/DPC%2B%2BFPGA/Tutorials/Features/loop_initiation_interval/README.md

# Additional details
the links should point to the docs:

[Xsclock](https://www.intel.com/content/www/us/en/develop/documentation/oneapi-fpga-optimization-guide/top/flags-attr-prag-ext/optimization-flags/spcfy-schd-fmax-trgt.html)
[Initiation Interval](https://www.intel.com/content/www/us/en/develop/documentation/oneapi-fpga-optimization-guide/top/flags-attr-prag-ext/loop-directives/ii-attribute.html)
Loop-carried dependencies -> I don't know where this was supposed to point.

SUGARXUST, 2022-09-22:
Hey @whitepau the links work fine for me!

bopeng1234, 2023-04-25:
OneSmpl_Team1 working on this, PR created

closed 2023-07-13 by jimmytwei