oneapi-src/oneAPI-samples #937: Is multi-gpu supported in oneapi?

kakulo, 2022-05-12:
I am running a few CNN/DNN examples within oneDNN, but it always run on a single Nvidia GPU. Is there a way to run on multiple Nivida GPUs? Is it currently supported or do I miss something? Thanks!

aice-support, 2022-07-27:
oneDNN supports computation & memory primitives.
For communication support such as all reduce among multiple devices, oneCCL supports those communications operations. 
However, you might need to use NCCL for those communication operations on NVidia GPUs.

yehudaorel, 2022-11-01:
> oneDNN supports computation & memory primitives. For communication support such as all reduce among multiple devices, oneCCL supports those communications operations. However, you might need to use NCCL for those communication operations on NVidia GPUs.

@kakulo 
Was that able to answer your question?

aice-support, 2022-11-16:
@kakulo 
Hope our response solves your problem.
We will close the issue for now, and feel free to reopen the issue if you still have problem.

closed 2022-11-16 by aice-support