oneapi-src/oneAPI-samples #1087: Minor edits after my DPC++ lab yesterday

plvlnisse, 2022-09-14:
# Summary
Clarifications to the lab exercises on essential OneAPI/DPC++
# URLs
a/DirectProgramming/DPC++/Jupyter/oneapi-essentials-training/03_DPCPP_Unified_Shared_Memory/Unified_Shared_Memory.ipynb
a/DirectProgramming/DPC++/Jupyter/oneapi-essentials-training/04_DPCPP_Sub_Groups/Sub_Groups.ipynb 
a/DirectProgramming/DPC++/Jupyter/oneapi-essentials-training/04_DPCPP_Sub_Groups/lab/sub_group_lab.cpp
# Additional details
These are merely some rewording to see if this is a way to provide feedback. There might be some more as I go along
[oneapi-essentials-training.patches.txt](https://github.com/oneapi-src/oneAPI-samples/files/9565344/oneapi-essentials-training.patches.txt)

hexu33, 2023-04-25:
OneSmpl_Team1 working on this

hexu33, 2023-04-25:
(OneSmpl_Team1)

@plvlnisse 

This issue will not occur in the latest version. This issue can be closed now.