oneapi-src/oneAPI-samples #1333: Shared memory allocation failure on FPGA

ManuelCostanzo, 2023-02-05:
Hi!

I'm using this example:
https://github.com/oneapi-src/oneAPI-samples/blob/master/DirectProgramming/C%2B%2BSYCL/DenseLinearAlgebra/vector-add/src/vector-add-usm.cpp

But when I run it in the DevClud I get:

Running on device: pac_a10 : Intel PAC Platform (pac_ee00000)
Vector size: 10000
Shared memory allocation failure.
u180336@s001-n088:~/oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/vector-add/src$ SYCL_DEVICE_FILTER=acc:1 ./aaaa3.fpga
Running on device: pac_a10 : Intel PAC Platform (pac_ee00000)
Vector size: 10000
Shared memory allocation failure.

Does USM work in FPGAs? Thank you very much

wangyon1, 2023-04-25:
OneSmpl_Team1 working on this

hexu33, 2023-04-25:
(OneSmpl_Team1)

@ManuelCostanzo 

SYCL USM is not support on A10 PAC CARD. 
SYCL USM allocations are only supported on FPGA boards that have a USM capable BSP (e.g. the Intel® FPGA PAC D5005 with Intel Stratix® 10 SX with USM support: intel_s10sx_pac:pac_s10_usm) or when targeting an FPGA family/part number.

Reference: https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/C%2B%2BSYCL_FPGA/Tutorials/GettingStarted/fpga_compile

closed 2023-08-12 by jimmytwei