oneapi-src/oneAPI-samples #1337: Unable to run sample SYCL program

DarshanRamakant, 2023-02-07:
# Summary
I have built SYCL compiler from the source. I was trying out the example given in the below documentation :
[https://intel.github.io/llvm-docs/GetStartedGuide.html#use-dpc-toolchain](url)

After building the compiler from source, I have updated my environment as suggested 
```
export PATH=$DPCPP_HOME/llvm/build/bin:$PATH
export LD_LIBRARY_PATH=$DPCPP_HOME/llvm/build/lib:$LD_LIBRARY_PATH
```
I could build the binary successfully, however on running that executable , I am getting below error 

Command used to generate binary 
`clang++ -fsycl simple_sycl.cpp -o syc
`

```
./syc
terminate called after throwing an instance of 'sycl::_V1::runtime_error'
  what():  No device of requested type available. -1 (PI_ERROR_DEVICE_NOT_FOUND)
Aborted (core dumped)
```

# Version
Tip commit of git 
3df87e20569ea63d0a74de525b6d19788dd8afca

# Environment
OS
Ubuntu 20.04.3 LTS

CPU information 
```
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   43 bits physical, 48 bits virtual
CPU(s):                          12
On-line CPU(s) list:             0-11
Thread(s) per core:              1
Core(s) per socket:              1
Socket(s):                       12
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           85
Model name:                      Intel(R) Xeon(R) Gold 5220R CPU @ 2.20GHz
Stepping:                        0
CPU MHz:                         2194.844
```

# Steps to reproduce
Command used to generate binary 
`clang++ -fsycl simple_sycl.cpp -o syc
`
./syc

# Observed behavior
I am suspecting missing runtime libraries.

hexu33, 2023-04-25:
OneSmpl_Team1 working on this

hexu33, 2023-04-25:
(OneSmpl_Team1)

@DarshanRamakant 

it's a runtime issue, you could try "clinfo -l" to check if the environment is configured correctly. For example:

$ clinfo -l
Platform #0: Intel(R) FPGA Emulation Platform for OpenCL(TM)
 `-- Device #0: Intel(R) FPGA Emulation Device
Platform #1: Intel(R) OpenCL
 `-- Device #0: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
Platform #2: Intel(R) OpenCL HD Graphics
 `-- Device #0: Intel(R) Iris(R) Xe Graphics [0x9a49]

vasistavinay, 2023-05-02:
I'm facing the exact same issue on my machine.

`$ sycl_clang++ -fsycl simple_sycl.cpp -o simple.o`
`$ ./hello.o`
```
terminate called after throwing an instance of 'sycl::_V1::runtime_error'
  what():  No device of requested type available. -1 (PI_ERROR_DEVICE_NOT_FOUND)
Aborted (core dumped)
```

Version:
`origin/sycl : 6a13e45c5abd3f3d74d81efba23e1578095afd67`

Config:
`python $DPCPP_HOME/llvm/buildbot/configure.py -o ~/builds --enable-esimd-emulator`

The following command returned nothing :
`clinfo -l`

Without the option `-l`, it returned this:
`Number of platforms                               0`

What should I do to make the code compile and run only on host (which has no device attached) ?

DarshanRamakant, 2023-05-03:
> I'm facing the exact same issue on my machine.
> 
> `$ sycl_clang++ -fsycl simple_sycl.cpp -o simple.o` `$ ./simple.o`
> 
> ```
> terminate called after throwing an instance of 'sycl::_V1::runtime_error'
>   what():  No device of requested type available. -1 (PI_ERROR_DEVICE_NOT_FOUND)
> Aborted (core dumped)
> ```
> 
> Version: `origin/sycl : 6a13e45c5abd3f3d74d81efba23e1578095afd67`
> 
> Config: `python $DPCPP_HOME/llvm/buildbot/configure.py -o ~/builds --enable-esimd-emulator`
> 
> The following command returned nothing : `clinfo -l`
> 
> Without the option `-l`, it returned this: `Number of platforms 0`
> 
> What should I do to make the code compile and run only on host (which has no device attached) ?

I am also getting the exact same output. Numberof platform is empty. 

Why dont you fix your "configure.py" to add the correct runtime dependencies ?

vasistavinay, 2023-05-03:
@DarshanRamakant by `fix your "configure.py"` do you mean
(a) I should supply proper options in my command line?
Or
(b) the script should be smart enough to detect device availability in my system?

vasistavinay, 2023-05-03:
I see a similar issue posted in Intel LLVM discussions :
https://github.com/intel/llvm/discussions/4493

DarshanRamakant, 2023-05-03:
> @DarshanRamakant by `fix your "configure.py"` do you mean (a) I should supply proper options in my command line? Or (b) the script should be smart enough to detect device availability in my system?

I was suggesting this to @hexu33 or any other developer who will be fixing this issue. I expect configure.py to take care of all the dependencies and installation. 

This is a hello-world example, it will be such a turn-off if this itself is not working.