oneapi-src/oneAPI-samples #1343: cmake does not recognize icpx as a valid compiler

tzachicohen, 2023-02-08:
# Summary
when running cmake configuration stage, I get the follwing error:

master) tcohen@tcohen:build $ cmake ..
-- The CXX compiler identification is IntelLLVM 2023.0.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx
-- Check for working CXX compiler: /opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx - broken
CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCXXCompiler.cmake:62 (message):
  The C++ compiler

    "/opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx"

  **is not able to compile a simple test program.**

  It fails with the following output:

    Change Dir: /home/tcohen/code/oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/simple-add/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_9dafe/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_9dafe.dir/build.make CMakeFiles/cmTC_9dafe.dir/build
    gmake[1]: Entering directory '/home/tcohen/code/oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/simple-add/build/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_9dafe.dir/testCXXCompiler.cxx.o
    /opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx    -MD -MT CMakeFiles/cmTC_9dafe.dir/testCXXCompiler.cxx.o -MF CMakeFiles/cmTC_9dafe.dir/testCXXCompiler.cxx.o.d -o CMakeFiles/cmTC_9dafe.dir/testCXXCompiler.cxx.o -c /home/tcohen/code/oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/simple-add/build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_9dafe
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_9dafe.dir/link.txt --verbose=1
    /opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx CMakeFiles/cmTC_9dafe.dir/testCXXCompiler.cxx.o -o cmTC_9dafe 
    /usr/bin/ld: cannot find -lstdc++: No such file or directory
    icpx: error: linker command failed with exit code 1 (use -v to see invocation)
    gmake[1]: *** [CMakeFiles/cmTC_9dafe.dir/build.make:100: cmTC_9dafe] Error 1
    gmake[1]: Leaving directory '/home/tcohen/code/oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/simple-add/build/CMakeFiles/CMakeTmp'
    gmake: *** [Makefile:127: cmTC_9dafe/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!

# Version
Latest OneAPI toolkit and samples. 

# Environment
Ubuntu 22.04

# Steps to reproduce
Install Tools kit according to Intel instructions: 
https://www.intel.com/content/www/us/en/developer/tools/oneapi/base-toolkit-download.html?operatingsystem=linux&distributions=aptpackagemanager

Download the samples and run cmake config stage.

# Observed behavior
cmake config stage fails

# Expected behavior
came config stage is successful.

hexu33, 2023-04-25:
OneSmpl_Team1 working on this

hexu33, 2023-04-25:
(OneSmpl_Team1)

@tzachicohen 

you might fix your "cannot find -lstdc++" error with install the C++ standard library, for example, "sudo apt install libstdc++6 libstdc++-11-dev" on ubuntu