oneapi-src/oneAPI-samples #1350: Can't build "Vector Add" Sample

vmayoral, 2023-02-13:
### Summary
Can't build [`Vector Add`](https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/C%2B%2BSYCL/DenseLinearAlgebra/vector-add) targeting `fgpa` on Linux.

*:warning: As an additional bit, it's non-intuitive to load the files generated by the build system selected into Quartus GUI. It'd be nice to get short instructions on how to load it all as a project into Quartus.*

### Version

- oneAPI: `v0.2.0-7-gab4eb7c822`
- quartus: `22.2.0 Build 94 06/08/2022 SC Pro Edition`

```bash
victor@accelerationrobotics:/media/victor/intel/oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/vector-add/build$ oneapi-cli version
v0.2.0-7-gab4eb7c822

victor@accelerationrobotics:/media/victor/intel/oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/vector-add/build$ quartus_cmd -v
Quartus Prime
Version 22.2.0 Build 94 06/08/2022 SC Pro Edition
Copyright (C) 2022  Intel Corporation. All rights reserved.
```

### Environment
Ubuntu Linux 22.04,  5.15.0-60-generic, x86_64

### Steps to reproduce
```bash
mkdir build
cd build; cmake ..
make fpga
Consolidate compiler generated dependencies of target vector-add-buffers.fpga
[ 50%] Linking CXX executable ../vector-add-buffers.fpga
aoc: Compiling for FPGA. This process may take several hours to complete.  Prior to performing this compile, be sure to check the reports to ensure the design will meet your performance targets.  If the reports indicate performance targets are not being met, code edits may be required.  Please refer to the oneAPI FPGA Optimization Guide for information on performance tuning applications for FPGAs.
Error: The board package being used only supports report generation/emulation/runtime, please use the board package shipped in the FPGA addon instead.
Error: Exiting.

llvm-foreach:
icpx: error: fpga compiler command failed with exit code 1 (use -v to see invocation)
make[3]: *** [src/CMakeFiles/vector-add-buffers.fpga.dir/build.make:97: vector-add-buffers.fpga] Error 1
make[2]: *** [CMakeFiles/Makefile2:263: src/CMakeFiles/vector-add-buffers.fpga.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:296: src/CMakeFiles/fpga.dir/rule] Error 2
make: *** [Makefile:215: fpga] Error 2
```

### Observed behavior
Error being thrown while invoking quartus. It's not obvious which device's being used from the tutorial but further looking into the source files, it appears that it's defaulting to `intel_a10gx_pac:pac_a10` (*I'd be nice to set this in the instructions, even if reduntant*).

### Expected behavior
FPGA bitstreams generated for target `FPGA_DEVICE`.

mohpan, 2023-02-14:
I am suspecting you might still face some dependency issues, if so you can try this direct command instead, 

[image]

vmayoral, 2023-02-15:
> As you guessed by default the target FPGA configuration file (SOF) is generated for PAC-platform (Acceleration card). For which the right BSP needs to "installed"/pointed-to. For your use case you can change the default target to Agilex/Arria10/Stratix10 family in general or any specific device part by using this switch while running the cmake command
cmake .. -DFPGA_DEVICE=agilex
Don't try any device less than Arria10 because, to compile SOF for low end devices (Max, CycleV etc) you need to install Quartus-Standard version. For all other high end devices you need to use Quartus-Pro (the one you have installed).

This just did it. Thanks a lot @mohpan!

For compleness, here's the recipe to build the `Vector Add` example with `Quartus Pro 2022.2` and targeting `agilex` boards:

```bash
source /opt/intel/oneapi/setvars.sh
mkdir build; cd build
cmake .. -DFPGA_DEVICE=agilex
make fpga
```

closed 2023-02-15 by vmayoral