oneapi-src/oneAPI-samples #1355: FPGA tests don't all work using Quartus

adrianjhpc, 2023-02-14:
# Summary
The board_test example (and other examples I've tested) from DirectProgramming/C++SYCL_FPGA/ReferenceDesigns fails on the part of the test that tries to extract the reference frequency using the `aocl binedit` functionality.

# Version
Report oneAPI Toolkit version and oneAPI Sample version or hash.

oneAPI version 2022.0.2
quartus version 20.4
oneAPI sample hash f6660eef7007d0acb7de1bce9c6e5844c6615a80

# Environment

Intel Stratix 10, Bittware N520 MX

# Steps to reproduce
```
cd ~/oneAPI-samples/DirectProgramming/C++SYCL_FPGA/ReferenceDesigns/board_test
mkdir build
cd build
cmake .. -DFPGA_BOARD=s10mx:p520_hpc_m210h_g3x16
make fpga
./board_test.fpga
```

# Observed behavior
When compiled and run the `board_test.fpga` executable fails, with this output at the end of the run:

```
*****************************************************************
*******************  Kernel Clock Frequency Test  ***************
*****************************************************************

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
```

Digging into the code it is because this functionality:
```
   std::string temp_cmd = "aocl binedit " + kBinaryName + " print .acl.quartus_report";
   const char *open_cmd_rep = temp_cmd.c_str();
```

Doesn't actually work. The executable (`board_test.fpga`) does not have anything to report when using `1aocl binedit board_test.fpga list`, i.e. it produces an empty output:

```
Sections in package file:
```

This can be fixed by changing the command to this:

```
   std::string temp_cmd = "aocl binedit " + kBinaryName + ".prj/base.aocx" + " print .acl.quartus_report";
   const char *open_cmd_rep = temp_cmd.c_str();
```
Which then can be used to extract the reference FPGA frequency.

I can commit this as a pull request.

# Expected behavior
The board_test to run without issues.

hexu33, 2023-04-25:
OneSmpl_Team1 working on this.

hexu33, 2023-04-25:
Hi, 
Please upgrade your OneAPI and FPGA add-on version to the latest and have a try.