oneapi-src/oneAPI-samples #1368: profile_utils.py:164 causes warning message

shuangpe, 2023-02-20:
# Summary
The profile_utils.py causes following annoying warning messages. Please figure out a way to eliminate it.
`profile_utils.py:164: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame.
Try using .loc[row_indexer,col_indexer] = value instead

See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  data['time'] = data['time'].astype(float)`

# Version
Python 3.10.6
Pandas 1.5.3

# Environment
Ubuntu 22.04.1 LTS

# Steps to reproduce
python3 profile_utils.py <your onednn verbose log file>

# Observed behavior
The annoying warning messages presented above

# Expected behavior
No warning messages presented in the console output

sau0, 2023-04-26:
OneSmpl_Team3 working on this.

kumarva1, 2023-04-27:
(OneSmpl_Team3) -
@shuangpe
We tried running this code with various arguments, but we see no warnings coming.
PFB screenshot for the same:
[image]

PR #1620 (2023-05-11): [oneDNN][profile_utils.py] Warning message fix
# Existing Sample Changes
### Description
Fix for warning message output in python_utils script when passing oneDNN verbose logs with *info* lines.

Fixes  #1368 
Currently, when passing a oneDNN verbose log to python_utils script with info lines the script outputs warning messages to the user:

```profile_utils.py:354: SettingWithCopyWarning:
A value is trying to be set on a copy of a slice from a DataFrame.
Try using .loc[row_indexer,col_indexer] = value instead

See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/ [... cut]