oneapi-src/oneAPI-samples #1370: IntelGfxDbgPackage did not load correctly in vs2019

NoWait126, 2023-02-21:
1 My computer configuration
OS: win10
processor: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz   2.42 GHz
graphics card: Intel(R) Iris(R) Xe Graphics
memory: 16G
vistual studio: 2019
cmake:3.26-rc3

2 install
I downloaded `w_BaseKit_p_2023.0.0.25940_offline.exe` and followed [this](https://www.intel.com/content/www/us/en/develop/documentation/get-started-with-intel-oneapi-base-windows/top/before-you-begin.html#before-you-begin_GUID-18E1F76E-93F2-4E99-8357-1C710A1DD130) to install, and I installed graphics drivers

3 run
I followed [this](https://www.intel.com/content/www/us/en/develop/documentation/get-started-with-intel-oneapi-base-windows/top/run-a-sample-project-using-an-ide.html) to run

4 What did I get
After I opened `vs2019`  I got error dialog saying `IntelGfxDbgPackage did not load correctly`,  I closed this error dialog  and selected  `Extensions > Intel > Browse an Intel oneAPI Sample > vector add`,  but found the `Debug > Start Without Debugging` is gray and un-clickable.

5 My question
Does my computer support `oneapi`? If so, why cannot I run the sample, do I forget install or config sth?

hexu33, 2023-04-25:
OneSmpl_Team1 working on this

hexu33, 2023-04-25:
(OneSmpl_Team1)

@NoWait126 
This issue has been addressed.

https://community.intel.com/t5/Intel-oneAPI-Base-Toolkit/IntelGfxDbgPackage-did-not-load-correctly-in-vs2019/td-p/1457860