oneapi-src/oneAPI-samples #1376: Libraries/oneDPL/ examples throw a lot of DEPRECATED warnings with oneAPI 2023.0.0

rasenjop, 2023-02-22:
# Summary
Just to point out that oneDPL 2022 version provided with oneAPI 2023 needs to be updated.

# Version
oneAPI 2023.0.0 on Ubuntu 20.04

# Steps to reproduce
cd oneAPI-samples/Libraries/oneDPL/maxloc_reductions
make

Plenty of  DEPRECATED warnings arise. For example:

/opt/intel/oneapi/dpl/2022.0.0/linux/include/oneapi/dpl/pstl/hetero/dpcpp/parallel_backend_sycl.h:727:91: warning: 'local' is deprecated: use `local_accessor` instead [-Wdeprecated-declarations]

/opt/intel/oneapi/dpl/2022.0.0/linux/include/oneapi/dpl/pstl/hetero/dpcpp/execution_sycl_defs.h:390:27: warning: 'get_info<sycl::info::kernel_device_specific::max_sub_group_size>' is deprecated: Use the overload without the second parameter [-Wdeprecated-declarations]

hexu33, 2023-04-25:
OneSmpl_Team1 working on this.

hexu33, 2023-04-25:
(OneSmpl_Team1)

@rasenjop 

tried latest code with oneAPI 2023.1 release, no DEPRECATED warnings now.