oneapi-src/oneAPI-samples #1412: 11th Gen Intel GPU not supported. Why?

irudkin, 2023-03-08:
# Summary
The example program crashes/aborts with following error:
./matrix_mul_mkl
Device: Intel(R) Iris(R) Xe Graphics [0x9a49]
Problem size:  A (600x1200) * B (1200x2400)  -->  C (600x2400)
Launching oneMKL GEMM calculation...
An exception occurred: oneapi::mkl::oneapi::mkl::blas::dgemm: unsupported device: Intel(R) Iris(R) Xe Graphics [0x9a49]
make: *** [GNUmakefile:8: run] Error 1

# Version
Report oneAPI Toolkit version 2023.0.0. and oneAPI Sample version downloaded 3/8/2023

# Environment
OS: Ubuntu 22.04.5 LTS 64bit
Platform: Latitude 7320, 16Gb RAM, Intel i71185G7, (11th Gen) GPU Mesa Intel Xe TGL Gt2

# Steps to reproduce
Move to the root directory of the example and run make.

# Observed behavior
Crash

# Expected behavior
Program exit gracefully and report why the GPU is not suitable not supported.

skiselev, 2023-04-05:
The GPU offload for Ice Lake platform (GEN11 graphics) is supported. You'll need to install the legacy GPU drivers:
https://dgpu-docs.intel.com/releases_legacy/legacy-m815_20220830.html
The GPU offload for EHL platform (also GEN11 graphics) is not supported, and never was supported. It would be nice though to add a platform support matrix to these samples

bopeng1234, 2023-04-25:
(OneSmpl_Team1)

@irudkin
I have confirmed it runs well in my i5-1135G7, which use the same GPU with yours, I am on ubuntu 22.04 and oneAPI 2023.1.
```
Libraries/oneMKL/matrix_mul_mkl$ ./matrix_mul_mkl
Device: Intel(R) Iris(R) Xe Graphics [0x9a49]
Problem size:  A (600x1200) * B (1200x2400)  -->  C (600x2400)
Launching oneMKL GEMM calculation...
Performing reference calculation...
Results are accurate.
```