oneapi-src/oneAPI-samples #1413: Label for -DFPGA_DEVICE in "DirectProgramming/C++SYCL_FPGA/Tutorials/GettingStarted/fpga_compile" example

jdayana, 2023-03-09:
Hi, I am having some trouble with compiling the fpga_compile sample on Linux.

According to the README.MD, I should generate the `Makefile` by running:
`cmake .. -DFPGA_DEVICE=<board-support-package>:<board-variant>`

However, I am not sure what my board's -DFPGA_DEVICE name is. May I receive some assistance regarding this matter? 
I am using the Intel Agilex IA-840F FPGA board and I have downloaded and installed the SDK and Card Support Package (CSP) for this board.

Below is my output when I tried compiling using the default settings.

```
xinmatrix@xinmatrix:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile$ mkdir build
xinmatrix@xinmatrix:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile$ cd build
xinmatrix@xinmatrix:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build$ cmake ..
-- The CXX compiler identification is IntelLLVM 2022.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/intel/oneapi/compiler/2022.1.0/linux/bin/dpcpp - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- FPGA_BOARD was not specified.                    
Configuring the design to run on the default FPGA board intel_a10gx_pac:pac_a10 (Intel(R) PAC with Intel Arria(R) 10 GX FPGA).                     
Please refer to the README for information on board selection.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build
```
```
xinmatrix@xinmatrix:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build$ make fpga_emu
[ 50%] Building CXX object src/CMakeFiles/fpga_compile.fpga_emu.dir/fpga_compile.cpp.o
[100%] Linking CXX executable ../fpga_compile.fpga_emu
[100%] Built target fpga_compile.fpga_emu
[100%] Built target fpga_emu
xinmatrix@xinmatrix:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build$ make report
[ 50%] Building CXX object src/CMakeFiles/fpga_compile_report.a.dir/fpga_compile.cpp.o
[100%] Linking CXX executable ../fpga_compile_report.a
[100%] Built target fpga_compile_report.a
[100%] Built target report
```
```
xinmatrix@xinmatrix:~/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build$ make VERBOSE=1 fpga
/home/xinmatrix/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -S/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile -B/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build --check-build-system CMakeFiles/Makefile.cmake 0
make  -f CMakeFiles/Makefile2 fpga
make[1]: Entering directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
/home/xinmatrix/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -S/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile -B/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build --check-build-system CMakeFiles/Makefile.cmake 0
/home/xinmatrix/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_progress_start /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build/CMakeFiles 2
make  -f CMakeFiles/Makefile2 src/CMakeFiles/fpga.dir/all
make[2]: Entering directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
make  -f src/CMakeFiles/fpga_compile.fpga.dir/build.make src/CMakeFiles/fpga_compile.fpga.dir/depend
make[3]: Entering directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
cd /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build && /home/xinmatrix/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_depends "Unix Makefiles" /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/src /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build/src /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build/src/CMakeFiles/fpga_compile.fpga.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
make  -f src/CMakeFiles/fpga_compile.fpga.dir/build.make src/CMakeFiles/fpga_compile.fpga.dir/build
make[3]: Entering directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
[ 50%] Building CXX object src/CMakeFiles/fpga_compile.fpga.dir/fpga_compile.cpp.o
cd /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build/src && /opt/intel/oneapi/compiler/2022.1.0/linux/bin/dpcpp   -Wall  -fintelfpga -MD -MT src/CMakeFiles/fpga_compile.fpga.dir/fpga_compile.cpp.o -MF CMakeFiles/fpga_compile.fpga.dir/fpga_compile.cpp.o.d -o CMakeFiles/fpga_compile.fpga.dir/fpga_compile.cpp.o -c /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/src/fpga_compile.cpp
[100%] Linking CXX executable ../fpga_compile.fpga
cd /home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build/src && /home/xinmatrix/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_link_script CMakeFiles/fpga_compile.fpga.dir/link.txt --verbose=1
/opt/intel/oneapi/compiler/2022.1.0/linux/bin/dpcpp -fintelfpga -Xshardware -Xsboard=intel_a10gx_pac:pac_a10  CMakeFiles/fpga_compile.fpga.dir/fpga_compile.cpp.o -o ../fpga_compile.fpga 
aoc: Compiling for FPGA. This process may take several hours to complete.  Prior to performing this compile, be sure to check the reports to ensure the design will meet your performance targets.  If the reports indicate performance targets are not being met, code edits may be required.  Please refer to the oneAPI FPGA Optimization Guide for information on performance tuning applications for FPGAs.
Error: The board package being used only supports report generation/emulation/runtime, please use the board package shipped in the FPGA addon instead.
Error: Exiting.

llvm-foreach: 
dpcpp: error: fpga compiler command failed with exit code 1 (use -v to see invocation)
make[3]: *** [src/CMakeFiles/fpga_compile.fpga.dir/build.make:97: fpga_compile.fpga] Error 1
make[3]: Leaving directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
make[2]: *** [CMakeFiles/Makefile2:208: src/CMakeFiles/fpga_compile.fpga.dir/all] Error 2
make[2]: Leaving directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
make[1]: *** [CMakeFiles/Makefile2:241: src/CMakeFiles/fpga.dir/rule] Error 2
make[1]: Leaving directory '/home/xinmatrix/oneAPI-samples/DirectProgramming/DPC++FPGA/Tutorials/GettingStarted/fpga_compile/build'
make: *** [Makefile:189: fpga] Error 2

```

Thanks!

dj-park, 2023-03-17:
Did you take a look at [this issue](https://github.com/oneapi-src/oneAPI-samples/issues/1350)?

jdayana, 2023-04-04:
I did, but was prompted with an error. I have not tried it again on the Linux OS after posting my issue.

Currently, I am using the Windows OS. And when I try to compile and build, I get the following: 
```
Microsoft Windows [Version 10.0.19044.2728]
(c) Microsoft Corporation. All rights reserved.

C:\Windows\system32>"C:\Program Files (x86)\Intel\oneAPI\setvars.bat"
:: initializing oneAPI environment...
   Initializing Visual Studio command-line environment...
   Visual Studio version 17.5.3 environment configured.
   "C:\Program Files\Microsoft Visual Studio\2022\Community\"
   Visual Studio command-line environment initialized for: 'x64'
:  advisor -- latest
:  compiler -- latest
:  dal -- latest
:  debugger -- latest
:  dev-utilities -- latest
:  dnnl -- latest
:  dpcpp-ct -- latest
:  dpl -- latest
:  intelpython -- latest
:  ipp -- latest
:  ippcp -- latest
:  mkl -- latest
:  mpi -- latest
:  tbb -- latest
:  vpl -- latest
:  vtune -- latest
:: oneAPI environment initialized ::

C:\Windows\System32>cd "C:\Users\jdayana\Downloads\oneAPI-samples-master_new\DirectProgramming\C++SYCL_FPGA\Tutorials\GettingStarted\fpga_compile\part1_cpp"

C:\Users\jdayana\Downloads\oneAPI-samples-master_new\DirectProgramming\C++SYCL_FPGA\Tutorials\GettingStarted\fpga_compile\part1_cpp>mkdir b1

C:\Users\jdayana\Downloads\oneAPI-samples-master_new\DirectProgramming\C++SYCL_FPGA\Tutorials\GettingStarted\fpga_compile\part1_cpp>cd b1

C:\Users\jdayana\Downloads\oneAPI-samples-master_new\DirectProgramming\C++SYCL_FPGA\Tutorials\GettingStarted\fpga_compile\part1_cpp\b1>cmake -G "NMake Makefiles" .. -DFPGA_DEVICE=agilex
-- The CXX compiler identification is IntelLLVM 2022.1.0 with MSVC-like command-line
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Intel/oneAPI/compiler/latest/windows/bin/icx.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Ignoring FPGA_DEVICE: agilex, not applicable
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/jdayana/Downloads/oneAPI-samples-master_new/DirectProgramming/C++SYCL_FPGA/Tutorials/GettingStarted/fpga_compile/part1_cpp/b1

C:\Users\jdayana\Downloads\oneAPI-samples-master_new\DirectProgramming\C++SYCL_FPGA\Tutorials\GettingStarted\fpga_compile\part1_cpp\b1>nmake fpga_emu

Microsoft (R) Program Maintenance Utility Version 14.35.32216.1
Copyright (C) Microsoft Corporation.  All rights reserved.

[ 50%] Building CXX object src/CMakeFiles/vector_add.fpga_emu.dir/vector_add.cpp.obj
In file included from C:\Users\jdayana\Downloads\oneAPI-samples-master_new\DirectProgramming\C++SYCL_FPGA\Tutorials\GettingStarted\fpga_compile\part1_cpp\src\vector_add.cpp:1:
In file included from C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\include\iostream:9:
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\include\yvals_core.h(807,1): error: static_assert failed "Error in C++ Standard Library usage."
_EMIT_STL_ERROR(STL1000, "Unexpected compiler version, expected Clang 15.0.0 or newer.");
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\include\yvals_core.h(462,5): note: expanded from macro '_EMIT_STL_ERROR'
    static_assert(false, "Error in C++ Standard Library usage.")
    ^             ~~~~~
1 error generated.
NMAKE : fatal error U1077: 'C:\PROGRA~2\Intel\oneAPI\compiler\latest\windows\bin\icx.exe @C:\Users\jdayana\AppData\Local\Temp\nmEC81.tmp' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\bin\HostX64\x64\nmake.exe" -s -f src\CMakeFiles\vector_add.fpga_emu.dir\build.make /nologo -SL                 src\CMakeFiles\vector_add.fpga_emu.dir\build' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\bin\HostX64\x64\nmake.exe" -s -f CMakeFiles\Makefile2 /nologo -SL                 src\CMakeFiles\fpga_emu.dir\all' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.35.32215\bin\HostX64\x64\nmake.exe" -s -f CMakeFiles\Makefile2 /nologo -S                  fpga_emu' : return code '0x2'
Stop.
```
May I know how I can solve this issue?

Note: When I first try to compile, I get the same issue as the one described [here](https://github.com/oneapi-src/oneAPI-samples/pull/1168). So I changed "CMAKE_FORCE_CXX_COMPILER (icx-cl IntelDPCPP)" to "set(CMAKE_CXX_COMPILER icx)". 

Thanks.

yuguen-intel, 2023-04-06:
You are getting the icx-cl error because your are using an older version of the compiler that does not support icx-cl.
If you want to continue using this version of the compiler, you need to use the version of the code samples that matches the compiler. 
The top level readme explains how to fetch a tagged version of the samples that corresponds to your version of the compiler.

Regarding the BSP you need to specify to target your FPGA board, I'd suggest you reach out to your board vendor.
I know they have another board which is compatible with the tool (the DE10 Agilex), so they should be able to help you figure out if your board has a BSP to you could use for these compiles.

yuguen-intel, 2023-08-22:
Closing this issue as this is expected behavior.

closed 2023-08-22 by yuguen-intel