oneapi-src/oneAPI-samples #1512: Bug on Devcloud?

ManuelCostanzo, 2023-04-01:
# Summary
Hi, I'm trying to compile for FPGAs in the Intel Devcloud. The problem is that I'm getting an error when I try to compile for hardware, it works well for emulator.

# Steps to reproduce
qsub -l nodes=1:fpga_compile:ppn=2 -I
```
source /opt/intel/oneapi/setvars.sh
cd ~/oneAPI-samples/DirectProgramming/C++SYCL_FPGA/Tutorials/GettingStarted/fast_recompile/src
icpx -fsycl -fintelfpga -fsycl-link=image kernel.cpp -o dev_image.a -Xshardware  -Xstarget=/opt/intel/oneapi/intel_a10gx_pac:pac_a10
```

# Observed behavior
The log I'm getting is:

```
u180336@s001-n063:~/oneAPI-samples/DirectProgramming/C++SYCL_FPGA/Tutorials/GettingStarted/fast_recompile/src$ icpx -fsycl -fintelfpga -fsycl-link=image kernel.cpp -o dev_image.a -Xshardware  -Xstarget=/opt/intel/oneapi/intel_a10gx_pac:pac_a10
aoc: Compiling for FPGA. This process may take several hours to complete.  Prior to performing this compile, be sure to check the reports to ensure the design will meet your performance targets.  If the reports indicate performance targets are not being met, code edits may be required.  Please refer to the oneAPI FPGA Optimization Guide for information on performance tuning applications for FPGAs.
Error (23035): Tcl error: 
Error (23031): Evaluation of Tcl script build/entry.tcl unsuccessful
Error: Quartus Prime Shell was unsuccessful. 2 errors, 0 warnings
For more details, full Quartus compile output can be found in files quartuserr.tmp and quartus_sh_compile.log.
Error: Compiler Error, not able to generate hardware

llvm-foreach: 
icpx: error: fpga compiler command failed with exit code 1 (use -v to see invocation)
```

Someone knows how to deal with this? Thank you very much!

robertszafa, 2023-04-03:
This issue has been ongoing since December afaik:
https://community.intel.com/t5/Intel-DevCloud/FPGA-Builds-Broken-on-DevCloud/td-p/1439447

Someone from devcloud support advised using the FPGA runtime nodes for compilation, which works, but blocks other users for hours from accessing the FPGA to run binaries:
https://community.intel.com/t5/Intel-DevCloud/Re-FPGA-Builds-Broken-on-DevCloud/m-p/1471112

I think the problem started after the 2023 update, which seems to have changed how quartus is bundled with oneapi.

ManuelCostanzo, 2023-04-03:
Hi @robertszafa, neither in the runtime nodes is working, I'm getting:

aoc: Compiling for FPGA. This process may take several hours to complete.  Prior to performing this compile, be sure to check the reports to ensure the design will meet your performance targets.  If the reports indicate performance targets are not being met, code edits may be required.  Please refer to the oneAPI FPGA Optimization Guide for information on performance tuning applications for FPGAs.
Error: Failed to open quartus_sh_compile.log
Error: /tmp/kernel-7d27d6-2c881a doesn't exist

it's a different error... I entered with:

qsub -l nodes=1:fpga_runtime:stratix10:ppn=2 -I