oneapi-src/oneAPI-samples #1513: FPGA simulator flow not working since the 2023 update.

robertszafa, 2023-04-03:
# Summary
My FPGA simulation flow broke after updating to intel-basekit-2023. I had a working setup with dpcpp 2022.2.0, Quartus prime pro 21.4, and Questa 21.4. Now I am able to compile for simulation but the SYCL runtime doesn't find a simulation device:
`Caught a SYCL host exception:
No device of requested type available. Please check https://software.intel.com/content/www/us/en/develop/articles/intel-oneapi-dpcpp-system-requirements.html -1 (PI_ERROR_DEVICE_NOT_FOUND)
terminate called after throwing an instance of 'sycl::_V1::runtime_error'
  what():  No device of requested type available. Please check https://software.intel.com/content/www/us/en/develop/articles/intel-oneapi-dpcpp-system-requirements.html -1 (PI_ERROR_DEVICE_NOT_FOUND)
Aborted`

I am able to start the `vsim` simulator and `quartus` on its own without problems, so its not a license issue. I also have all environment variables set (https://www.intel.com/content/www/us/en/docs/programmable/683472/21-4/environment-variables.html).

The issue persists even when I re-installed everything: a fresh Ubuntu 20.04 install, intel-basekit-2023.1.0, Quartus 21.4 and Questa 21.3. 
(I also tried using Quartus 19.2 and modelsim 19.2 on ubuntu 18 but I get different errors there during compilation.)

Could someone please give me some pointers on how to get the simulation flow going with the Questa Started Edition? I would really appreciate any help. I tried A LOT of things over the last week without any success. I had it all working before updating to 2023, so I know it's possible.

# Version
oneAPI Toolkit version  2023.1.0 (2023.1.0.20230320) 
oneAPI Sample hash 5be94fc7db259dafd54e8aedfcc99b39dc025cfc

# Environment
Ubuntu 20.04
Quartus 21.4 and Questa Starter Edition 21.3 installed from (https://www.intel.com/content/www/us/en/software-kit/706104/intel-quartus-prime-pro-edition-design-software-version-21-4-for-linux.html)

# Steps to reproduce
Following the GettingStarted guide:
`export CL_CONTEXT_COMPILER_MODE_INTELFPGA=3 && export CL_CONTEXT_MPSIM_DEVICE_INTELFPGA=1`
`cd DirectProgramming/C++SYCL_FPGA/Tutorials/GettingStarted/fpga_compile/part4_dpcpp_lambda_buffers/src`
`icpx -fsycl -fintelfpga -DFPGA_SIMULATOR -I../../../../include vector_add.cpp -Xssimulation -Xsghdl -o vector_add_sim.fpga_sim`
`# vector_add_sim.fpga_sim compiles succesfully`
`./vector_add_sim.fpga_sim`
`# Error: -1 (PI_ERROR_DEVICE_NOT_FOUND)`

# Observed behavior
Runtime error: ` -1 (PI_ERROR_DEVICE_NOT_FOUND)`

# Expected behavior
`vsim` runs the simulation and a `simulation_raw.json` file is produced with cycle counts. 

# Side note
I tried to go back to dpcpp 2022.1.0 (2022.2.0 is no longer available) but SYCL still cannot find the simulator device. In the past, I installed Quartus 21.4 using an apt package (https://apt.repos.intel.com/oneapi) like intel-oneapi-intelfpgadpcpp-a10gx-s10sx-quartus-21.4. This placed quartus in `/opt/intel/oneapi/intelfpgadpcpp/2022.2.0/QuartusPrimePro/21.4/quartus`. This apt package is no longer available. I tried installing `intel-oneapi-intelfpgadpcpp-a10gx-s10sx-quartus` but this installs Quartus 19.2 which fails to compile for simulation. It seems like the intelfpgadpcpp apt packages are outdated because a lot of them have either broken dependencies or error out with `Mirror sync in progress?`.

robertszafa, 2023-04-04:
This issue was solved by installing Quartus 22.4 (bundled with Questa Started Edition) on Ubuntu 22.

### My exact setup:
1. Ubuntu 22.04 and intel-basekit-2023.1.0
3. Download and install [Intel® Quartus® Prime Pro Edition Software (Device support not included)](https://www.intel.com/content/www/us/en/software-kit/764010/intel-quartus-prime-pro-edition-design-software-version-22-4-for-linux.html) and [Intel® Arria® 10 Device Support](https://www.intel.com/content/www/us/en/software-kit/764010/intel-quartus-prime-pro-edition-design-software-version-22-4-for-linux.html). The Quartus installer includes Questa Starter Edition.
4. Add to bashrc: 
```bash
# Added by installers
export QSYS_ROOTDIR=".../intelFPGA_pro/22.4/qsys/bin"
source /opt/intel/oneapi/setvars.sh > /dev/null 2>&1
# Added manually
export LM_LICENSE_FILE="..."
export PATH="$PATH:.../intelFPGA_pro/22.4/questa_fse/bin"
export PATH="$PATH:.../intelFPGA_pro/22.4/quartus/bin"
```
5. Install libraries missing on Ubuntu 22:
```bash
sudo apt-get install libncurses5 libxrender1 libxtst6 libxi6 libgconf-2-4
```
6. Now you can compile and run under simulation: 
```bash
icpx -fsycl -fintelfpga -DFPGA_SIMULATOR -I../../../../include vector_add.cpp -Xssimulation -Xstarget=Arria10 -Xsghdl -o vector_add_sim.fpga_sim
CL_CONTEXT_MPSIM_DEVICE_INTELFPGA=1 ./vector_add_sim.fpga_sim
```

HxJi, 2023-07-25:
Hi Robert @robertszafa, could you share which version of oneAPI sample codes you used? We also try to run the simulation with the same setup as yours but encounter the following error:

vector_add_sim.fpga_sim: src/hls_cosim_ipc_socket.cpp:249: void IPCSocketMaster::connect(): Assertion `0 && "IPCSocketMaster::connect() call to accept() failed"' failed. Aborted (core dumped)

Any suggestion is appreciated. Thanks

robertszafa, 2023-07-27:
The error indicates an issue with the Questa license. This link has some info on how to get a free license for Questa:
https://www.intel.com/content/www/us/en/docs/programmable/683472/22-1/and-software-license.html