oneapi-src/oneAPI-samples #1535: platform_designer broken with oneAPI 2023.0 after changes for 2023.1

vmayoral, 2023-04-10:
https://github.com/oneapi-src/oneAPI-samples/commit/876a9de0ee518daa6e19141b1542078ca23988e4 breaks `platform_designer`:

#### ❌  `master` right now (https://github.com/oneapi-src/oneAPI-samples/commit/876a9de0ee518daa6e19141b1542078ca23988e4):

```bash
cd DirectProgramming/C++SYCL_FPGA/Tutorials/Tools/experimental/platform_designer
cd add-oneapi && mkdir build && cd build
cmake .. -DFPGA_DEVICE=agilex
make fpga_ip_export
[ 33%] To compile manually:
/opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx -I../../../../../../include -fsycl -fintelfpga -Wall -qactypes -DFPGA_HARDWARE -c ../src/add.cpp -o CMakeFiles/fpga_ip_export.dir/src/add.cpp.o

To link manually:
/opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx -fsycl -fintelfpga -Xshardware -Xstarget=agilex -fsycl-link=early -fsycl-device-code-split=per_kernel -o add.fpga_ip_export CMakeFiles/fpga_ip_export.dir/src/add.cpp.o
[ 33%] Built target displayExportCompileCommands
[ 66%] Building CXX object CMakeFiles/fpga_ip_export.dir/src/add.cpp.o
/media/victor/intel/oneAPI-samples/DirectProgramming/C++SYCL_FPGA/Tutorials/Tools/experimental/platform_designer/add-oneapi/src/add.cpp:50:21: error: no member named 'fpga_selector_v' in namespace 'sycl::ext::intel'; did you mean 'sycl::cpu_selector_v'?
    auto selector = sycl::ext::intel::fpga_selector_v;
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    sycl::cpu_selector_v
/opt/intel/oneapi/compiler/2023.0.0/linux/bin-llvm/../include/sycl/device_selector.hpp:110:19: note: 'sycl::cpu_selector_v' declared here
__SYCL_EXPORT int cpu_selector_v(const device &dev);
                  ^
1 error generated.
make[3]: *** [CMakeFiles/fpga_ip_export.dir/build.make:76: CMakeFiles/fpga_ip_export.dir/src/add.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:204: CMakeFiles/fpga_ip_export.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:211: CMakeFiles/fpga_ip_export.dir/rule] Error 2
make: *** [Makefile:176: fpga_ip_export] Error 2
```

#### ✅  Before https://github.com/oneapi-src/oneAPI-samples/commit/876a9de0ee518daa6e19141b1542078ca23988e4 (e.g. while at https://github.com/oneapi-src/oneAPI-samples/commit/1aa9741bd06957ba70b41639a01f6c095f7ceaa6):
```bash
cd DirectProgramming/C++SYCL_FPGA/Tutorials/Tools/experimental/platform_designer
cd add-oneapi && mkdir build && cd build
cmake .. -DFPGA_DEVICE=agilex
make fpga_ip_export
[ 33%] To compile manually:
/opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx -I../../../../../../include -fsycl -fintelfpga -Wall -qactypes -DFPGA_HARDWARE -c ../src/add.cpp -o CMakeFiles/fpga_ip_export.dir/src/add.cpp.o

To link manually:
/opt/intel/oneapi/compiler/2023.0.0/linux/bin/icpx -fsycl -fintelfpga -Xshardware -Xstarget=agilex -fsycl-link=early -fsycl-device-code-split=per_kernel -o add.fpga_ip_export CMakeFiles/fpga_ip_export.dir/src/add.cpp.o
[ 33%] Built target displayExportCompileCommands
[ 66%] Building CXX object CMakeFiles/fpga_ip_export.dir/src/add.cpp.o
[100%] Linking CXX executable add.fpga_ip_export
[100%] Built target fpga_ip_export
```

Things work perfectly fine before this commit with oneAPI 2023.0. Ping @whitepau @yuguen-intel.

vmayoral, 2023-04-10:
When updating oneAPI to 2023.1 (`intel-basekit-2023.1.0`) previous error is gone.

Encouraging to drop a note on this regard and/or to recuperate the changes introduced by @whitepau.

yuguen-intel, 2023-04-11:
Hey @vmayoral, this is working as designed
The state of the master branch is to be used with the latest version of the compiler (it matches the latest github tag, currently "2023.1.0").
So the current master branch is expected to work with the 2023.1 release of the compiler.

The code samples are not backward/forward compatible, and you can only expect the samples to work if you checkout the github tag associated with your compiler release. This is all documented in the code samples top level README file.

The development branch (where the commit you are referring to belongs) is Intel's development branch in between two releases.
The `platform_designer` sample was not part of the 2023.0 samples, but was introduced in the development branch for it to make it to the 2023.1 release.
As this sample was anticipated by some external users, @whitepau made a "hack" for it to work with the 2023.0 release while waiting for the 2023.1 release to drop.
Now that 2023.1 dropped, the hack is no longer needed nor desired.
Indeed, we do not want to have release specific code macros -> we instead expect the users to fetch the correct version of the code samples that matches their compiler.

closed 2023-04-11 by yuguen-intel

vmayoral, 2023-04-13:
Makes sense, thanks.