oneapi-src/oneAPI-samples #1563: PipeArray example is not runnable on Intel devcloud

Arthur-Chang016, 2023-04-18:
# Summary
It seems that the tutorial in https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/C%2B%2BSYCL_FPGA/include
is not consistent with the latest dpc++ version

# Version
'icpx --version'
Intel(R) oneAPI DPC++/C++ Compiler 2023.0.0 (2023.0.0.20221201)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /glob/development-tools/versions/oneapi/2023.0.1/oneapi/compiler/2023.0.0/linux/bin-llvm
Configuration file: /glob/development-tools/versions/oneapi/2023.0.1/oneapi/compiler/2023.0.0/linux/bin-llvm/../bin/icpx.cfg

# Environment
Intel devcloud

# Steps to reproduce
run the tutorial
https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/C%2B%2BSYCL_FPGA/Tutorials/DesignPatterns/pipe_array
on Intel devcloud 

```
mkdir build
cd build
make
```

# Observed behavior
[image]

# Expected behavior
should be able to run the PipeArray example

hexu33, 2023-04-25:
OneSmpl_Team1 working on this.

hexu33, 2023-04-25:
(OneSmpl_Team1)

@Arthur-Chang016

sycl::ext:: intel:: fpga_emulator_selector_v can be found at oneAPI 2023.1 release, Current DevCloud automatic installed 2023.0 release as you provide, so you could either try latest oneAPI release, or select the corresponding oneAPI-samples tag.