oneapi-src/oneAPI-samples #1730: Error compiling oneDNN examples: DSO missing from command line

jjfumero, 2023-06-15:
# Summary

I am compiling the getting_started example from oneDNN: 
https://github.com/oneapi-src/oneAPI-samples/tree/master/Libraries/oneDNN/getting_started 

But I get an error during compilation. Perhaps I am missing something. I attach the logs and how to reproduce.

```bash
$ make
[  1%] Building CXX object bin/CMakeFiles/sycl-interop-usm-cpp.dir/sycl_interop_usm.cpp.o
[  3%] Linking CXX executable sycl-interop-usm-cpp
/usr/bin/ld: /tmp/icx-1a4810/sycl_interop_usm-410ec9.o: undefined reference to symbol '_ZSt20__throw_length_errorPKc@@GLIBCXX_3.4'
/usr/bin/ld: /usr/lib64/libstdc++.so.6: error adding symbols: DSO missing from command line
icx: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/CMakeFiles/sycl-interop-usm-cpp.dir/build.make:100: bin/sycl-interop-usm-cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:162: bin/CMakeFiles/sycl-interop-usm-cpp.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

# Version

oneAPI: 2023.1.0
oneAPI samples: b86bc9000b5ce89a143f9a1a8ec23743dbe78e7e 

# Environment

OS: RHEL 9.2 (Linux Kernel 5.14.0-284.11.1.el9_2.x86_64)
GCC/G++ : 11.3.1 20221121 (Red Hat 11.3.1-4)

# Steps to reproduce

```bash
git clone git@github.com:oneapi-src/oneAPI-samples.git
cd oneAPI-samples/Libraries/oneDNN/getting_started/
mkdir build
cd build
. /opt/intel/oneapi/setvars.sh
cmake ..
make 
```

# Observed behavior

```bash
make
[  1%] Building CXX object bin/CMakeFiles/sycl-interop-usm-cpp.dir/sycl_interop_usm.cpp.o
[  3%] Linking CXX executable sycl-interop-usm-cpp
/usr/bin/ld: /tmp/icx-4628a6/sycl_interop_usm-58e7f4.o: undefined reference to symbol '_ZSt20__throw_length_errorPKc@@GLIBCXX_3.4'
/usr/bin/ld: /usr/lib64/libstdc++.so.6: error adding symbols: DSO missing from command line
icx: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/CMakeFiles/sycl-interop-usm-cpp.dir/build.make:100: bin/sycl-interop-usm-cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:162: bin/CMakeFiles/sycl-interop-usm-cpp.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

# Expected behavior

Compile without errors.

jinz2014, 2023-07-11:
Do you need to specify the compilers ?

mkdir build
cd build
CC=GCC CXX=g++ cmake ..
make

HaoZeke, 2024-05-09:
You need `CXX=icpx` for Intel C++ and `icx` for the data compiler.