oneapi-src/oneAPI-samples #1796: default branch setting

jinz2014, 2023-07-12:
Could you set 'development' as the default branch for your repository ?  The Intel LLVM sets "sycl" as the default branch.

jimmytwei, 2023-08-12:
I don't have the permissions to set the ``development`` branch as the default branch.

The ``master`` branch is default since it is the branch that is compatible with the latest released version of the Intel oneAPI toolkits. The ``development`` branch is implied by its name - code that is still in development and that is not guaranteed to work with the latest released version of the Intel oneAPI toolkits. That's why we don't make it default... to prevent users from cloning the repo and possibly getting source code that doesn't work.

I will close out this issue. If you have any further questions, please reopen it.

closed 2023-08-12 by jimmytwei

jinz2014, 2023-08-13:
Okay.