oneapi-src/oneAPI-samples #1805: Simulation encounters error

HxJi, 2023-07-25:
# Summary
I tried to run the vector_add example under DirectProgramming/C++SYCL. The emulation runs well and the simulation can be compiled successfully. However, the simulation failed to run. 

# Version
oneAPI Toolkit version: 2023.01
oneAPI Sample version:  `5be94f`

# Environment
OS: Ubuntu 22.04
Quartus Prime Pro 22.4
Questa Starter 22.4

# Steps to reproduce
cd oneAPI-samples/DirectProgramming/C++SYCL/DenseLinearAlgebra/vector-add/
mkdir build
cd build
cmake ..
make fpga_sim
CL_CONTEXT_MPSIM_DEVICE_INTELFPGA=1 ./vector-add-buffers.fpga_sim

# Observed behavior
The simulation failed and returned the following message: 

```
The simulation process encountered an error and has aborted.
Error: The simulator's process ended unexpectedly. Check oneAPI-samples/DirectProgramming/C++SYCL_FPGA/Tutorials/GettingStarted/fpga_compile/part4_dpcpp_lambda_buffers/src/vector_add_sim.fpga_sim.prj/transcript.log for details.
vector_add_sim.fpga_sim: src/hls_cosim_ipc_socket.cpp:249: void IPCSocketMaster::connect(): Assertion `0 && "IPCSocketMaster::connect() call to accept() failed"' failed.
Aborted (core dumped)
```

# Expected behavior
The simulation runs well.

jimmytwei, 2023-08-12:
@yuguen-intel Any ideas?

yuguen-intel, 2023-08-16:
What is the content of the transcript.log file?

HxJi, 2023-08-23:
Thanks for getting back to me. I am updating Quartus to a newer version 23.2 and rerun the simulation. Will upload the log file when it is done.

yuguen-intel, 2024-11-25:
As there was no update I'm closing the case

closed 2024-11-25 by yuguen-intel

YazanMPSUT, 2025-04-07:
Facing the the same issue currently with Quartus 2025.1 and Questa FSE.