oneapi-src/oneAPI-samples #1951: How can hardened CPUs be programmed on the Intel dev cloud available FPGAs?

mrakgr, 2023-10-07:
By those I mean the Stratix, Arria and the Agilex lines which have integrated quad core ARM CPUs. I want to do a reinforcement learning related project to see how much of a speedup implementing everything on FPGA would bring. For that, I want to implement the game parts on the CPU, but it has to be on the FPGA otherwise the latency issue of moving data over the PCIe bus would kill the whole thing as it would on a GPU.

I am still relatively new to FPGAs and completely new to oneAPI, so I am wondering if there are any tutorials or examples how this could be done?

mrakgr, 2023-10-07:
Sorry if I wasn't clear. By this I mean, I want to somehow target the CPU parts of the FPGA using SYCL, and have it call out to the PL parts using the queue.

yuguen-intel, 2024-11-25:
This is not something you can do with the oneAPI runtime

closed 2024-11-25 by yuguen-intel