oneapi-src/oneAPI-samples #2069: want an example for target an explicit FPGA board variant and BSP

shiyi23, 2023-12-01:
The https://github.com/oneapi-src/oneAPI-samples/tree/master/DirectProgramming/C%2B%2BSYCL_FPGA/ReferenceDesigns/cholesky
say 
```
......

Alternatively, you can target an explicit FPGA board variant and BSP by using the following command:

cmake .. -DFPGA_DEVICE=<board-support-package>:<board-variant> -DIS_BSP=1
You will only be able to run an executable on the FPGA if you specified a BSP.

......
```

Can you provide a specific example of this `cmake .. -DFPGA_DEVICE=<board-support-package>:<board-variant> -DIS_BSP=1` command? Thanks~

umairsiddiqui-digitek, 2024-02-14:
https://ofs.github.io/ofs-2023.3-2/hw/common/user_guides/oneapi_asp/ug_oneapi_asp/#24-build-and-install-oneapi-asp

yuguen-intel, 2024-11-25:
Closing assuming the provided example was enough

closed 2024-11-25 by yuguen-intel